espruino/Espruino #1629: SyntaxError in logical expressions fail for some cases

MaBecker, 2019-03-07:
add ( logical expressions ) for this test files 
* test_array_access.js
* test_array_big_indices.js
* test_function_length.js
* test_object_proto.js 
* test_string_match.js

gfwilliams, 2019-03-08:
How are you uploading? Do you have your own tool?

The issue is the usual Espruino gotcha - if you copy/paste into the REPL then it'll execute after a newline unless there are more open brackets than closed.

If you upload with the Espruino CLI then those line endings get turned into Alt+Enter and it should all work great - so I'm not sure if this is something that we should really be tweaking tests for.

If you want a quick hack to get your own code uploading, just surround everything in `{` and `}`.

MaBecker, 2019-03-08:
upload like described in #1490

MaBecker, 2019-03-08:
but tested in console and WebIDE upload too.

gfwilliams, 2019-03-08:
Thanks - it was a CLI/WebIDE issue - if you use the latest from GitHub it should work now.

It's a good one to fix as I bet that's been tripping people up.

closed 2019-03-08 by gfwilliams