espruino/Espruino #1631: findings for branch experimental_compact_vars

MaBecker, 2019-03-08:
build for  ESP8266_4MB

negative values  in vars

```
>x = -1
=2047
>x = -10
=2038
>x = 0
=0
>x =1024
=1024
>x = -1024
=1024
```

```
>x = -2048
=-2048
>x = -4096
=-4096
```

```
>-1 -2
=-3
```

gfwilliams, 2019-03-11:
This looks like a bug in the `experimental_compact_vars` branch to do with `jsvGetFirstChildSigned`

gfwilliams, 2019-03-11:
Good spot though - thanks!

MaBecker, 2019-03-11:
So this is the line, any hints how to fix it?

https://github.com/espruino/Espruino/blob/master/src/jsvar.h#L272

gfwilliams, 2019-03-11:
Just done.

MaBecker, 2019-03-11:
Just tested, works

MaBecker, 2019-03-12:
Not shure if all those test should be successful for a ESP8266 baord

results of test_language.js

* [Uncaught ReferenceError: "1" is not defined](https://github.com/espruino/Espruino/blob/master/tests/test_language.js#L85)

* [Uncaught ReferenceError: "varGlobal" is not defined](https://github.com/espruino/Espruino/blob/master/tests/test_language.js#L800)

* 354 out of 401 passed

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v01.65 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:1024/1024, manuf 0x1c chip 0x3016
>Uncaught ReferenceError: "1" is not defined
 at line 1 col 22
['a',undefined,'c'][1]
                     ^
at line 4 col 21
  } else testsPass++;
                    ^
in function "shouldBe" called from line 1 col 47
shouldBe("['a',undefined,'c'][1]", "undefined");
                                              ^
ERROR: propnames[0] should be '0' but found 0
ERROR: propnames[1] should be '1' but found 1
ERROR: propnames[2] should be '2' but found 2
ERROR: u.charCodeAt(1) should be 4660 but found 52
ERROR: Number.MAX_VALUE.toString(2).length should be 23 but found 69
ERROR: delete Array.prototype should be false but found true
ERROR: var i; i should be undefined but found 1
ERROR: val should be 11 but found undefined
ERROR: val should be 12 but found undefined
ERROR: val should be 13 but found undefined
ERROR: val should be 14 but found function () {}
ERROR: val should be 15 but found 0
ERROR: delete nonexistant; should be true but found false
ERROR: Function declaration takes effect at entry: value has type string , not:function
ERROR: Decl not yet overwritten: value has type string , not:function
ERROR: function decls have no execution content: value is function () {} , not:3
ERROR: function decls have no execution content: value is function () {} , not:3
ERROR: function decls have no execution content: value is function () {} , not:3
ERROR: String()+Math.E should be '2.718281828459045' but found 2.71828182845
ERROR: String()+Math.LN2 should be '0.6931471805599453' but found 0.69314718055
ERROR: String()+Math.LN10 should be '2.302585092994046' but found 2.30258509299
ERROR: String()+Math.LOG2E should be '1.4426950408889634' but found 1.44269504088
ERROR: String()+Math.LOG10E should be '0.4342944819032518' but found 0.43429448190
ERROR: String()+Math.PI should be '3.141592653589793' but found 3.14159265358
ERROR: String()+Math.SQRT1_2 should be '0.7071067811865476' but found 0.70710678118
ERROR: String()+Math.SQRT2 should be '1.4142135623730951' but found 1.41421356237
>ERROR: isNegativeZero(Math.round(negativeZero)) should be true but found false
ERROR: Math.round(99999999999999999999.99) should be 100000000000000000000 but found 9223372036854775808
ERROR: Math.round(-99999999999999999999.99) should be -100000000000000000000 but found -9223372036854775808
ERROR: Math.pow(2, 32) >> one should be 0 but found -1
ERROR: Math.pow(2, 32) << one should be 0 but found -2
ERROR: Math.pow(2, 32) >>> one should be 0 but found 2147483647
ERROR: ' ' <= 0 should be true but found false
ERROR: g should be 'foo' but found baz
Uncaught ReferenceError: "varGlobal" is not defined
 at line 1 col 6
if (!varGlobal)
     ^
ERROR: varInFunction() should be true but found false
ERROR: varInVarList() should be true but found false
ERROR: varInBlock() should be true but found false
ERROR: varInIf() should be true but found false
ERROR: varInElse() should be true but found false
ERROR: varInDoWhile() should be true but found false
ERROR: varInWhile() should be true but found false
ERROR: varInFor() should be true but found false
ERROR: varInWith() should be true but found false
ERROR: varInCase() should be true but found false
ERROR: varInForInitExpr() should be true but found false
Uncaught ReferenceError: "varGlobal" is not defined
 at line 1 col 1
varGlobal
^
at line 4 col 21
  } else testsPass++;
                    ^
in function "shouldBe" called from line 1 col 26
shouldBe("varGlobal", "1");
                         ^
354 out of 401 passed
```

gfwilliams, 2019-03-12:
test_language definitely fails even on the Linux build. I guess the real question is whether there's a difference between the results of this branch and the master branch?

MaBecker, 2019-03-13:
Wow even some fail with node v8.9.4 on mac os x

```
node /Espruino/repos/Espruino/tests/test_language.js 
ERROR: Number.MAX_VALUE.toString(2).length should be 23 but found 1024
ERROR: val should be 13 but found undefined
ERROR: a = 1; delete a; should be true but found false
ERROR: Function declaration takes effect at entry: value has type undefined , not:function
ERROR: Decl already overwritten: value has type function , not:number
ERROR: Decl already overwritten: value has type function , not:number
395 out of 401 passed
```

MaBecker, 2019-04-16:
Next Steps:

* compare master test results with ecv branch results
* try some sensors
* and displays

MaBecker, 2019-06-28:
Test results:

master
```
cat master/test.summary.result 
Number of test files  299 /Espruino/repos/testing/esp8266_test.lst
Number of syntax errors  5
Number of reference errors  22
Number of result = true  207
Number of result = false  6
Number of result = 1  69
Number of result = 0  5
```

experimental_compact_vars

```
cat experimental_compact_vars/test.summary.result 
Number of test files  299 /Espruino/repos/testing/esp8266_test.lst
Number of syntax errors  5
Number of reference errors  21
Number of result = true  207
Number of result = false  6
Number of result = 1  69
Number of result = 0  5
```

MaBecker, 2019-06-28:
[esp8266_test.lst.txt](https://github.com/espruino/Espruino/files/3339944/esp8266_test.lst.txt)

gfwilliams, 2019-06-28:
Thanks, this is really reassuring! So there is one difference with "Number of reference errors" - do we know what the test is?

MaBecker, 2019-06-28:
This branch is easier to handle than trying to get the exception handler to work!

branch experimental_compact_vars

```
>process.env
={
  VERSION: "2v01.4",
  GIT_COMMIT: "e6146752",
  BOARD: "ESP8266_4MB",
  FLASH: 0, RAM: 81920,
  SERIAL: "84f3eb23-ae4c",
  CONSOLE: "Serial1",
  MODULES: "Flash,Storage,hea" ... "r,crypto,neopixel",
  EXPTR: 1073643636 }

>process.memory();
={ free: 1535, usage: 65, total: 1600, history: 9,
  gc: 0, gctime: 2.651 }

>require('ESP8266').getState();
={
  sdkVersion: "2.2.1(6ab97e9)",
  cpuFrequency: 160, freeHeap: 19720, maxCon: 10,
  flashMap: "4MB:1024/1024",
  flashKB: 4096,
  flashChip: "0xef 0x4016"
 }
```

MaBecker, 2019-06-28:
> Thanks, this is really reassuring! So there is one difference with "Number of reference errors" - do we know what the test is?

yep there is one difference, have to check

MaBecker, 2019-06-29:
Finding `test_regexp.js` is failing on experimental_compact_vars

AFAIK there are some regexp issues 

https://github.com/espruino/Espruino/search?q=regexp&type=Commits

gfwilliams, 2019-07-01:
Oh right, nice - so it's just that I haven't merged with master branch in a while? Can you try that test now and see if it works?

MaBecker, 2019-07-01:
> Oh right, nice - so it's just that I haven't merged with master branch in a while? Can you try that test now and see if it works? I just merged in all the recent changes.

Yes, it works, thanks for that merge.

MaBecker, 2019-07-01:
update 
```
>require('ESP8266').getState();
={
  sdkVersion: "2.2.1(6ab97e9)",
  cpuFrequency: 160, freeHeap: 20472, maxCon: 10,
  flashMap: "4MB:1024/1024",
  flashKB: 4096,
  flashChip: "0xef 0x4016"
 }
```

gfwilliams, 2019-07-01:
Perfect - so every test you've run performs identically now.

Maybe I should think about merging this... I guess only downside is maybe reduced performance, but it should give a lot more efficiency on devices with > 1024 vars.

MaBecker, 2019-07-01:
Open on my test list:  

* displays
  * oled (i2c)
  * lcd (i2c)
* sensors
  * any recommendation?
* other
  * neopixel

anything else comes to your mind?

gfwilliams, 2019-07-01:
Nope... As far as I'm concerned, if the main test cases work then it's 99.9% guaranteed that the hardware libs won't notice any difference either - but obviously it's good to be sure because I regret it every time I think that :)

MaBecker, 2019-09-13:
Successfully tested: 

- [oled over i2c ](http://www.espruino.com/SSD1306)
- neopixel 
- [DHT22](http://www.espruino.com/DHT22)
- [DS18B20](http://www.espruino.com/DS18B20)
- [BME280](http://www.espruino.com/BME280)

MaBecker, 2019-09-13:
**vars: 1700
freeHeap: 17208** 

require('ESP8266').getState();
={
  sdkVersion: "2.2.1(6ab97e9)",
  cpuFrequency: 160, freeHeap: 17208, maxCon: 10,
  flashMap: "4MB:1024/1024",
  flashKB: 4096,
  flashChip: "0x85 0x6014"
 }

process.memory()
={ free: 1663, usage: 37, total: 1700, history: 14,
  gc: 0, gctime: 2.852 }

MaBecker, 2019-09-14:
ready to go 😉

MaBecker, 2019-09-15:
> Oh right, nice - so it's just that I haven't merged with master branch in a while? Can you try that test now and see if it works? I just merged in all the recent changes.

please merge again

gfwilliams, 2019-09-16:
Just done. Running through the diffs again this looks pretty good - I just need to put some time aside to test it on the other Espruino boards as right now this'll affect those too.

MaBecker, 2020-03-28:
Can you please merge again

Have you thought about taking this life, as you know it free much space :)

gfwilliams, 2020-10-29:
Just merged it again.

I was just looking at merging this into master, but it really bumps up the code size so it's going to be a bit of work to get Espruino Original and boards like micro:bit working with it. It'll take a bit more thought

MaBecker, 2020-10-29:
I was using it for ESP8266 1MB boards with no problems.

gfwilliams, 2020-10-29:
Yes, but they have 1MB of flash, not 256k :)

closed 2021-11-09 by MaBecker