espruino/Espruino #1634: ESP8266: Constructor argument is not passed while exporting

xargon180, 2019-03-17:
Hi the code bellow returns:
undefined
NaN

Expecting:
10
15

The code without require("Storage").write("answer",\` ............   \`); wrapper works fine.

require("Storage").write("answer", \`
  class TestClass {
    constructor(test) {
       print(test); // test is undefined!!!
      _test = test;
    }
    setTest(test) {
      print(test); // works as expected
      _test = test;
    }
    getResult() {
      return _test + 5;
    }
  }

exports.get = function() {
  return new TestClass(10);
}; \` );
print(require("answer").get().getResult());

gfwilliams, 2019-03-18:
Thanks for the report, but which version of Espruino are you running this on? I'm pretty sure this was something that got fixed a while back: https://github.com/espruino/Espruino/issues/1576

xargon180, 2019-03-18:
Hi gfwilliams thanks for your quick response. 
I use 2v01 for ESP8266. 
https://www.espruino.com/binaries/espruino_2v01_esp8266_4mb_combined_4096.bin

Fixing the bug is not a big priority for me. I have changed my code anyway based on your ES5 module template. Just wanted to let you know.

gfwilliams, 2019-03-18:
Thanks - just tested this.

It fails on the 2v01 release, but this one has already been fixed - so if you try a cutting edge build (or 2v02 when released) it'll be sorted :)

closed 2019-03-19 by gfwilliams