espruino/Espruino #1642: Second level require gives error but works

akaztp, 2019-04-09:
With the Typescript bootstrap project https://github.com/stasberkov/espruino-ts-quickstart, I setup these 3 files:

app.ts:
```
import { fnA } from './lib-a';
// import { fnB } from './lib-b';
fnA();
// fnB();
```
lib-a.ts:
```
import { fnB } from './lib-b';
export function fnA() {
  console.log('Here fnA');
  fnB();
}
```
lib-b.ts:
```
export function fnB() {
  console.log('Here fnB');
}
```

Sending this to Espruino 2v.01 results in:
```
>Uncaught Error: Module ./lib-b not found
 at line 1 col 152
...r lib_b_1=require('./lib-b');exports.fnA=fnA;
                              ^
>Here fnA
Here fnB
```

Uncommenting those two lines in `app.ts`, works fine and with no error:
```
>Here fnA
Here fnB
Here fnB
```

The `bundle.js` produced (with the comments back on `app.ts`) is:
```
Modules.addCached("./lib-a","'use strict';function fnA(){console.log('Here fnA'),lib_b_1.fnB();}Object.defineProperty(exports,'__esModule',{value:!0});var lib_b_1=require('./lib-b');exports.fnA=fnA;");
Modules.addCached("./lib-b","'use strict';function fnB(){console.log('Here fnB');}Object.defineProperty(exports,'__esModule',{value:!0}),exports.fnB=fnB;");
"use strict";

var lib_a_1 = require("./lib-a");
// import { fnB } from './lib-b';
lib_a_1.fnA();
// fnB();
```

gfwilliams, 2019-04-09:
This looks like an ordering issue - the espruino-ts-quickstart tool is sending (and hence loading) the module lib-a *before* the module lib-b that it uses. It's not too surprising that it can't be found.

If you just swap the two lines at the top:

```
Modules.addCached("./lib-b","'use strict';function fnB(){console.log('Here fnB');}Object.defineProperty(exports,'__esModule',{value:!0}),exports.fnB=fnB;");
Modules.addCached("./lib-a","'use strict';function fnA(){console.log('Here fnA'),lib_b_1.fnB();}Object.defineProperty(exports,'__esModule',{value:!0});var lib_b_1=require('./lib-b');exports.fnA=fnA;");
```

Then it works great. Please could you report it on https://github.com/stasberkov/espruino-ts-quickstart as it's an issue with those tools - not Espruino itself.

closed 2019-04-09 by gfwilliams

akaztp, 2019-04-09:
Hi Gordon, thank you for your fast answer (and for the amazing work on Espruino, I just found out a couple of weeks ago, amazing!)

Yes, I've spotted that earlier but it is the same thing. In fact, please realize that the code works fine, all 
`require` do work, only the error message is strange!

In fact I just retried swapping those two lines and added a new `console.log('hello');` to make sure I was seeing the updated code. I'll copy/paste into here:
```
Modules.addCached("./lib-b","'use strict';function fnB(){console.log('Here fnB');}Object.defineProperty(exports,'__esModule',{value:!0}),exports.fnB=fnB;");
Modules.addCached("./lib-a","'use strict';function fnA(){console.log('Here fnA'),lib_b_1.fnB();}Object.defineProperty(exports,'__esModule',{value:!0});var lib_b_1=require('./lib-b');exports.fnA=fnA;");
"use strict";

var lib_a_1 = require("./lib-a");
// import { fnB } from './lib-b';
lib_a_1.fnA();
// fnB();
console.log('hello');
```

Results in (and just after power on):
```

>Uncaught Error: Module ./lib-b not found
 at line 1 col 152
...r lib_b_1=require('./lib-b');exports.fnA=fnA;
                              ^
>Here fnA
Here fnB
hello
>
```

And the order of the `Modules.addCached()` calls might not matter if that second argument is evaluated only when the module is required and not when that `addCached()` is called.
I've searched for the source code of this `addCached()` function, but could not find it, nor using github find nor google!

Also strange is, if I require the `lib-b` *also* right at the top level, then the error message is gone, as I depicted in the first post.

gfwilliams, 2019-04-10:
I just tried the code you pasted above on a board and it works perfectly, without errors:

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v01.102 (c) 2018 G.Williams
>Here fnA
Here fnB
hello
> 
```

You sure you don't already have the broken code saved somewhere?

>  the order of the Modules.addCached() calls might not matter if that second argument is evaluated only when the module is required

The second argument *is* evaluated when it's uploaded. It's a memory saving measure - on a memory-constrained device you don't want two copies of a module hanging around (the texture version and the in-memory parsed version).

If you did `require('./lib-b')` inside a function call you wouldn't have the issue. It's because it's being done in global scope.

> I've searched for the source code of this addCached() function

Just find it in the reference: http://www.espruino.com/Reference#l_Modules_addCached

Then click the `=>` right after the title. It'll take you direct to the source: https://github.com/espruino/Espruino/blob/master/src/jswrap_modules.c#L236