espruino/Espruino #1643: STM32F3DISCOVERY not working

swineone, 2019-04-12:
STM32F3DISCOVERY is not working with the latest version of Espruino.

I'm opening the issue just to post a pull request with a fix.

swineone, 2019-04-12:
The fix is in #1644.

swineone, 2019-04-12:
Sorry for the mess, not very familiar with GitHub. I suppose the proper way is to leave this open until it is closed by the maintainers.

MaBecker, 2019-04-13:
Hi, 
a nice way is to include keyword "fix  #number" in the PR title.
Then the issue gets closed when the PR is merged. 

Sample: https://github.com/espruino/Espruino/pull/1627

gfwilliams, 2019-04-15:
Hi - thanks for the PR. 

Which bit of the build fails though, and how? Or do you get a firmware that doesn't work?

Because I just did a build here and it works fine for me - so it could just be something to do with how your build system is configured.

swineone, 2019-04-15:
Sorry, it didn't occur to me to explain what's going on in depth.

The firmware does build. However, when it is loaded, a hard fault handler occurs. I traced this to an attempt to initialize an MCU pin with an invalid value of `JSH_PORT_NONE` from the `pinInfo` array, so the address calculation of `port - JSH_PORTA` in `stmPort()` produces an invalid address.

It turns out that this MCU skips certain pins (e.g. PF3, although both PF2 and PF4 exist). Rather than wiping these non-existent pins from the `pinInfo` array, they're left with invalid value entries. Looking at the `boards/STM32F3DISCOVERY.py' file, it turns out someone commented out the code that calls `pinutils.only_from_package()` to filter out these invalid pins, so that's why they were left in the array.

I tried putting the `pinutils.only_from_package()` call back in, but now running the script fails due to an attempt to access `pin["csv"]`, but this key doesn't exist. This is because when a fake pin is created in `pinutils.fill_gaps_in_pin_list()`, no `csv` dictionary entry is created. I then patched `pinutils.only_from_package()` to check for the existence of `csv` in the dictionary before trying to access it, as can be seen in the pull request's diff.

gfwilliams, 2019-04-15:
That's awesome - thanks for all your work debugging/fixing this, and the in-depth explanation. I just merged it.

Sorry about the query - I have to check though as sometimes I get PRs that aren't actually needed and could cause problems elsewhere.

Can you do a quick check and see if the pins that are after a missing pin (eg. `F4`) are actually usable? It might be you need this line as well: https://github.com/espruino/Espruino/blob/master/boards/PICO_R1_3.py#L45

PR #1644 (2019-04-15): Fix bug in pin_utils.py's only_from_package() function
Fix bug in pin_utils.py's only_from_package() function, which in turn fixes a hard fault handler in STM32F3DISCOVERY. Fixes #1643.
closed 2019-04-15 by gfwilliams