espruino/Espruino #1654: Software Serial TX and RX

gfwilliams, 2019-04-24:
Started here: http://forum.espruino.com/conversations/333049

Software serial RX works alone, as does TX. However together there are issues:

```
var s = new Serial();
s.setup(600,{rx:B3, tx:B4});
s.on('data', function (data) { print("<SoftSerial>"+data); });
setTimeout(function() {
//  s.print("H"); // fine
//  s.print("Hello"); // garbage
}, 500);
```

Only the last character sent seems to be received correctly - at any baud rate.

closed 2019-04-24 by gfwilliams

MaBecker, 2019-04-24:
Hmm 7 byte very small, is it possible to increase the buffer?

https://github.com/espruino/Espruino/blob/23665d05da49dcc98384744fc393ac5804d3377c/src/jsserial.c#L194

gfwilliams, 2019-04-25:
Just done, to 64 - it does come out of JsVars so it's not going to cause too many problems I guess.

Still 7 bytes @ 9600 baud means you have around 7ms of execution time to fill before you hit issues

commit 23665d05 (2019-04-24): Fix Software Serial receive when the frame has to be decoded inside the IRQ (fix #1654)