espruino/Espruino #1663: Modify NFC to work with SDK 15.3

gfwilliams, 2019-05-17:
As usual, Nordic have completely refactored part of their SDK. While we can patch 15.2, 15.3 lacks the old 'hal_nfc_*' functions.

It doesn't look too bad to modify though. Andreas says:

Looking at targets/nrf5x/bluetooth.c we just need to rewire the calls from the 
old api to the new one

  - hal_nfc_setup -> nrfx_nfct_init
  - hal_nfc_start -> nrfx_nfct_enable
  - hal_nfc_parameter_set -> nrfx_nfct_parameter_set
  - hal_nfc_parameter_get -> nrfx_nfct_nfcid1_default_bytes_get
  - hal_nfc_send -> nrfx_nfct_tx
  - hal_nfc_send_rsp -> nrfx_nfct_tx (assiming frame size is in bits)
  - hal_nfc_stop -> nrfx_nfct_disable
  - hal_nfc_done -> nrfx_nfct_uninit (not sure if this one is needed)

wilberforce, 2019-05-17:
I never understand why library writers do this sort of thing? What are they trying to *improve*?

If the params are the same you might be able to get away with #define-ing the old to the new, or using macros - that way you might be abke to support the old and new libs and #define the symbols as in a #ifdef

drandreas, 2019-05-18:
The initial API was more or less tailored to a read only Type 2 NFC Tag emulator. I had to patch the HAL at several places to add writable tag support.
With every SDK iteration the API offered more and more features. So our patches to the SDK got less and less.
However most of the quirks in the SDK (workarounds for silicon errors) had to be enabled at compile time.
I assume the API has matured enough now that they are switching to an universal API for many/all Nordic Chips.

drandreas, 2021-08-29:
Were you successful in upgrading to a newer SDK?

gfwilliams, 2021-08-31:
No - I had a very quick stab a while back but unfortunately at the time I was more concerned with getting BLE working with SDK15 so never put much effort in, and haven't got around to it since.

There are now some patches that get applied on top of the NFC libs for SDK15 at https://github.com/espruino/Espruino/tree/master/targetlibs/nrf5x_15/patches - but those do not apply cleanly.