espruino/Espruino #1665: Wrong APB1/APB2 clocks on STM32F4DISCOVERY

swineone, 2019-05-21:
According to the STM32F407VG datasheet (the STM32F4DISCOVERY's MCU), Table 14 (p. 77), fHCLK <= 168 MHz, fPCLK1 <= 42 MHz and fPCLK2 <= 84 MHz. Assuming, as seems to be the case, that fHCLK is configured by Espruino to 168 MHz, then the clock divider must be 4 for APB1 and 2 for APB2. However, the [current code](https://github.com/espruino/Espruino/blob/master/targets/stm32/jshardware.c#L1216) uses 2 for APB1 and 4 for APB2, therefore overclocking APB1 and underclocking APB2. This incorrect behavior was verified by running `peek32()` on the relevant RCC register.

A fix will follow shortly.

gfwilliams, 2019-05-21:
Thanks - out of interest does this cause any issues that you noticed? Or is it just not what's shown in the datasheet.

closed 2019-05-21 by gfwilliams

swineone, 2019-05-21:
I haven't suffered any hardware issues that I could reasonably attribute to this yet, but of course, it's a gamble to run things (i.e. every peripheral hooked up to APB1) at twice their rated speed. ST's process node probably has advanced to the point where this isn't an issue, but still, it's a gamble.

I only noticed this because I was playing around with configuring timer registers directly via `poke32()`, and I was getting pulses at twice the frequency that I expected, so I started checking the RCC clock tree configuration via `peek32()` and found it.

gfwilliams, 2019-05-21:
Thanks! As you say, it's good practice anyway :)

PR #1666 (2019-05-21): Fix APB1 overclocking on STM32F4DISCOVERY.
Fixes #1665.