espruino/Espruino #1668: Method fill() cause reboot for large Arrays, Uint8Array works fine.

MaBecker, 2019-05-26:
Works on Espruino Pico board, so this a ESP8266 issue.

Running on latest Espruino build 

  VERSION: "2v03.11",
  GIT_COMMIT: "f7b96f93",
  BOARD: "ESP8266_4MB",

Want to use Arrays , because they have lot more methods than Uint8Arrays like push and shift.

Error using array

https://github.com/espruino/Espruino/blob/master/src/jsvar.c#L737

https://github.com/espruino/Espruino/blob/master/src/jswrap_array.c#L863-L885

```
>a = new Array(3000);
=[ undefined, undefined, undefined, undefined, undefined,  ... undefined, undefined, undefined, undefined, undefined ]
>
>a.fill(0);
ASSERT(var && jsvHasRef(var)) FAILED AT src/jsvar.c:737
  #1[r1,l2] Object { }
---console end---
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
```

No error for using Uint8Array

```
>a = new Uint8Array(3000);
=new Uint8Array(3000)
>a.fill(0);
=new Uint8Array(3000)
> 
```

Any hint how to fix this?

gfwilliams, 2019-05-28:
How many variables are there in the ESP8266 build? Because what you're doing is going to allocate an Array that uses up 3000 variables - probably more than ESP8266 has available?

Having said that, it's probably a sign that there should be an 'out of memory' check inside `Array.fill`

I just tried this on a linux build with reduced variables and the same happens - so will see about adding the check

closed 2019-05-28 by gfwilliams

MaBecker, 2019-05-28:
Thanks!

So there is a difference using Array and Uint8Array as this was working without problems for Uint8Array

gfwilliams, 2019-05-28:
Yes - there's a really big difference :) http://www.espruino.com/Performance#espruino-stores-normal-arrays-and-objects-in-linked-lists

commit 1b88d2b0 (2019-05-28): Fix Array.fill crash if used to fill up all available memory (fix #1668)