espruino/Espruino #1672: Graphics.asBMP does not respect setRotation

fanoush, 2019-06-13:
https://github.com/espruino/Espruino/blob/87446715fa955f5ecb54330658321e24c4938ad7/libs/graphics/jswrap_graphics.c#L1475

When using setRotation(1,0) width and height is not swapped so only rectangular area is shown correctly and rest of image is blank on one side and cut off on other side.
This can be seen best in WebIDE via dump method
```
g=Graphics.createArrayBuffer(64,128,1)
g.setRotation(1,0)
g.drawString("Hello world 0123456789",0,0);
g.dump()
```
Following change fixes it for me (took inspiration from other methods in same source). I did not test image mirroring but that does not change dimensions so should hopefully work.
```
diff --git a/libs/graphics/jswrap_graphics.c b/libs/graphics/jswrap_graphics.c
index b8387027..63a16219 100644
--- a/libs/graphics/jswrap_graphics.c
+++ b/libs/graphics/jswrap_graphics.c
@@ -1472,10 +1472,12 @@ JsVar *jswrap_graphics_asBMP(JsVar *parent) {
     jsExceptionHere(JSET_ERROR, "asBMP/asURL only works on 1bpp/24bpp Graphics");
     return 0;
   }
-  int rowstride = (((gfx.data.width*gfx.data.bpp)+31) >> 5) << 2; // padded to 32 bits
+  int width = (gfx.data.flags & JSGRAPHICSFLAGS_SWAP_XY) ? gfx.data.height : gfx.data.width;
+  int height = (gfx.data.flags & JSGRAPHICSFLAGS_SWAP_XY) ? gfx.data.width : gfx.data.height;
+  int rowstride = (((width*gfx.data.bpp)+31) >> 5) << 2; // padded to 32 bits
   bool hasPalette = gfx.data.bpp==1;
   int headerLen = 14+ 12+ (hasPalette?6:0);
-  int l = headerLen + gfx.data.height*rowstride;
+  int l = headerLen + height*rowstride;
   JsVar *imgData = jsvNewFlatStringOfLength((unsigned)l);
   if (!imgData) return 0; // not enough memory
   unsigned char *imgPtr = (unsigned char *)jsvGetFlatStringPointer(imgData);
@@ -1486,10 +1488,10 @@ JsVar *jswrap_graphics_asBMP(JsVar *parent) {
   imgPtr[10]=(unsigned char)headerLen;
   // BITMAPCOREHEADER
   imgPtr[14]=12; // sizeof(BITMAPCOREHEADER)
-  imgPtr[18]=(unsigned char)gfx.data.width;
-  imgPtr[19]=(unsigned char)(gfx.data.width>>8);
-  imgPtr[20]=(unsigned char)gfx.data.height;
-  imgPtr[21]=(unsigned char)(gfx.data.height>>8);
+  imgPtr[18]=(unsigned char)width;
+  imgPtr[19]=(unsigned char)(width>>8);
+  imgPtr[20]=(unsigned char)height;
+  imgPtr[21]=(unsigned char)(height>>8);
   imgPtr[22]=1;
   imgPtr[24]=(unsigned char)gfx.data.bpp; // bpp
   if (hasPalette) {
@@ -1497,10 +1499,10 @@ JsVar *jswrap_graphics_asBMP(JsVar *parent) {
     imgPtr[27]=255;
     imgPtr[28]=255;
   }
-  for (int y=0;y<gfx.data.height;y++) {
-    int yi = gfx.data.height-(y+1);
+  for (int y=0;y<height;y++) {
+    int yi = height-(y+1);
     if (gfx.data.bpp==1) {
-      for (int x=0;x<gfx.data.width;) {
+      for (int x=0;x<width;) {
         unsigned int b = 0;
         for (int i=0;i<8;i++) {
           b = (b<<1)|(graphicsGetPixel(&gfx, (short)(x++), (short)y)&1);
@@ -1508,7 +1510,7 @@ JsVar *jswrap_graphics_asBMP(JsVar *parent) {
         imgPtr[headerLen + (yi*rowstride) + (x>>3) - 1] = (unsigned char)b;
       }
     } else {
-      for (int x=0;x<gfx.data.width;x++) {
+      for (int x=0;x<width;x++) {
         unsigned int c = graphicsGetPixel(&gfx, (short)x, (short)y);
         int i = headerLen + (yi*rowstride) + (x*(gfx.data.bpp>>3));
         imgPtr[i++] = (unsigned char)(c);

```

gfwilliams, 2019-06-14:
This looks great, thanks! Again, do you want to do a PR (which will give you proper credit), or shall I just apply the patch manually this end?

fanoush, 2019-06-14:
Whatever is easier for you. I can send PR - for both issues. As with the setFontBitmap bug happened before, you may have better way to fix this as I may have been missing larger context so I just did a patch what seem to work for me.

closed 2019-06-17 by gfwilliams