espruino/Espruino #1676: Add handler to RODATA branch

MaBecker, 2019-06-17:
Get a simple handler and add it.

MaBecker, 2019-06-17:
added handler from nodemcu/nodemcu-firmware and try to get it working

For now it is just constantly rebooting and thowing garbage

MaBecker, 2019-06-23:
```

Fatal exception (0): 
epc1=0x401072d0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
details to epc1=0x401072d0
```
grep 401072d espruino_esp8266_user1.lst 
401072d0 g     F .iram0.text	0000001d __wrap__xtos_set_exception_handler
401072b9:	000686        	j	401072d7 <__wrap__xtos_set_exception_handler+0x7>
401072c8:	050c67        	bnone	a12, a6, 401072d1 <__wrap__xtos_set_exception_handler+0x1>
401072d0 <__wrap__xtos_set_exception_handler>:
401072d0:	4539a0        	extui	a3, a10, 25, 5
401072d3:	443b      	addi.n	a4, a4, 3
401072d5:	0249      	s32i.n	a4, a2, 0
401072d7:	3108      	l32i.n	a0, a1, 12
401072d9:	10c112        	addi	a1, a1, 16
401072dc:	f00d      	ret.n
401072de:	120000        	excw
```

```
grep _xtos_set_exception_handler espruino_esp8266_user1.lst 
401072d0 g     F .iram0.text	0000001d __wrap__xtos_set_exception_handler
40000454 g       *ABS*	00000000 _xtos_set_exception_handler
401072b9:	000686        	j	401072d7 <__wrap__xtos_set_exception_handler+0x7>
401072c8:	050c67        	bnone	a12, a6, 401072d1 <__wrap__xtos_set_exception_handler+0x1>
401072d0 <__wrap__xtos_set_exception_handler>:
__wrap__xtos_set_exception_handler():
401072ee:	000106        	j	401072f6 <__wrap__xtos_set_exception_handler+0x26>
```

Any hints how to procced?

MaBecker, 2019-06-23:
Just to add: this is about 

249> Heap: 17320
251> Variables: 1600 @16ea = 25600bytes

gfwilliams, 2019-06-24:
So it's crashing in the code that sets the exception handler?

Just some random stabs in the dark, but:

* Sometimes devices complain if you try and set stuff up *after* other things are initialised. You could try setting the handler right at boot time.
* Is the handler in the correct code area so that it stays in RAM all the time? I guess it should have the same stuff set that IRQ handlers have

MaBecker, 2019-06-27:
>1+1
=2
>require('ESP8266').getState();
 30517> Fatal exception (3):
 30517> epc1=0x4000bdcb, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4026f5ab, depc=0x00000000
 30518> Fatal exception (3):
 30520> epc1=0x4000bdcb, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4026f5ab, depc=0x00000000
 30529> Fatal exception (3):
 30531> epc1=0x40100601, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4026f5ab, depc=0x00000000
.....
 30517> Fatal exception (3):
 30517> epc1=0x4000bdcb, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4026f5ab, depc=0x00000000
 30518> Fatal exception (3):
 30520> epc1=0x4000bdcb, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4026f5ab, depc=0x00000000
 30529> Fatal exception (3):
 30531> epc1=0x40100601, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4026f5ab, depc=0x00000000
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0

Look's like the handler is not doing his job, because there should be no output for Fatal exception(3)

MaBecker, 2019-06-27:
just pushed the latest version to https://github.com/MaBecker/Espruino/tree/ESP8266_RODATA

Feel free to jump in and help to get this handler working :)

MaBecker, 2022-06-08:
close it for now and reopen if someone likes to implement this.

closed 2022-06-08 by MaBecker

gfwilliams, 2022-06-08:
Shame, that would have really helped