espruino/Espruino #1679: add one more section to flash  with ld option --rename, only ESP8266_4MB

MaBecker, 2019-06-24:
add rename .rodata=.irom.literal, because CCFLAG -mforce-l32 is used for ESP8266_4MB boards.

freeHeap +2064 byte

MaBecker, 2019-08-27:
have to reverse --rename-section .rodata=.irom.literal

Some function calls cause reboot, like

* JSON.parse(undefined)
* Date.parse("Aug 9, 1995")
. . . .

closed 2019-08-28 by gfwilliams

gfwilliams, 2019-08-28:
Thanks for sorting this quickly, it's frustrating about those issues - I wonder why `-mforce-l32` isn't doing its thing

MaBecker, 2019-08-28:
.text is not 2**4 arranged like the others

````
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         0000088c  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00002584  3ffe8890  3ffe8890  00000970  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .bss          0000db20  3ffeae18  3ffeae18  00002ef8  2**4
                  ALLOC
  3 .irom0.text   000753d4  40201010  40201010  00009ea0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00006fa0  40100000  40100000  00002ef4  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
```

commit 4834652e (2019-06-28): ESP8266 more heap (fix #1679)

- a win of +2064 byte for freeHeap
- print out header details for  section 0. - 5.
commit 6ea4c0a2 (2019-06-28): Merge pull request #1681 from espruino/ESP8266_more_heap

ESP8266 more heap (fix #1679)
commit e0e7da88 (2019-08-27): reverse  (fix #1679)
commit 06ae1f36 (2019-08-28): Merge pull request #1692 from espruino/ESP8266_reverse_fix1679

reverse  (fix #1679)