espruino/Espruino #1687: ESP32 and DHT22 low memory

ApocalypseLinyx, 2019-08-14:
Hi. I'm use esp32-wroom-32 and DHT22 sensor with [DHT22.js](http://www.espruino.com/modules/DHT22.js) script.

I call the read function every 5 seconds and every time I get less and less memory, the log below as it was at the beginning of work was 1917, and after a while it was already 1887 and so on until there is an error.

What my problem?

My code:
 ``` 
setInterval(function() {
    this.dht.read((response) => {
      console.log(response)
      console.log(process.memory())
    });
  }, 5000);
```

My logs: 

 ```
____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v04 (c) 2019 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
{
  "raw": "100000010001011010000000011100001000100001",
  "rh": 111.4, "temp": 45 }
{ "free": 1917, "usage": 383, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.989 }
{
  "raw": "010000001000101100000000001110000100001111",
  "rh": 55.6, "temp": 22.5 }
{ "free": 1929, "usage": 371, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.954 }
{
  "raw": "010000001000110010000000001110000100010101",
  "rh": 56.2, "temp": 22.5 }
{ "free": 1926, "usage": 374, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.96 }
{
  "raw": "010000001000110100000000001110000100010111",
  "rh": 56.4, "temp": 22.5 }
{ "free": 1922, "usage": 378, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.956 }
{
  "raw": "010000001000110001000000001110001000010101",
  "rh": 56.1, "temp": 22.6 }
{ "free": 1919, "usage": 381, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.965 }
{
  "raw": "010000001000101000000000001110001000001100",
  "rh": 55.2, "temp": 22.6 }
{ "free": 1915, "usage": 385, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.958 }
{
  "raw": "010000001000101000000000001110001000001100",
  "rh": 55.2, "temp": 22.6 }
{ "free": 1912, "usage": 388, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.96 }
{
  "raw": "010000001000110001000000001110001000010101",
  "rh": 56.1, "temp": 22.6 }
{ "free": 1908, "usage": 392, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.964 }
{
  "raw": "010000001001010011000000001110001000110111",
  "rh": 59.5, "temp": 22.6 }
{ "free": 1905, "usage": 395, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.961 }
{
  "raw": "010000001001010111000000001110001000111011",
  "rh": 59.9, "temp": 22.6 }
{ "free": 1901, "usage": 399, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.967 }
{
  "raw": "010000001001010011000000001110001000110111",
  "rh": 59.5, "temp": 22.6 }
{ "free": 1898, "usage": 402, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.963 }
{
  "raw": "010000001001001001000000001110001000101101",
  "rh": 58.5, "temp": 22.6 }
{ "free": 1894, "usage": 406, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.969 }
{
  "raw": "010000001000111011000000001110001000011111",
  "rh": 57.1, "temp": 22.6 }
{ "free": 1887, "usage": 413, "total": 2300, "history": 179,
  "gc": 0, "gctime": 1.966 }
```

MaBecker, 2019-08-16:
use a simple code like this to reproduce the memory leak: 

```
var dht = require("DHT22").connect(<used pin>);

setInterval(function() {
    dht.read(function (response) {
        console.log(response)
        console.log(process.memory());
});}, 5000);
```

ApocalypseLinyx, 2019-08-16:
Create thread on the forum:
http://forum.espruino.com/conversations/337313/

@MaBecker, me use this simple code and memory low after some time

MaBecker, 2019-08-16:
ok, just check the module with a ESP8266  without memory leaks.

The module is using some calls like 

`digitalWrite(ht.pin, 0);`

`pinMode(ht.pin,"output");`

`pinMode(ht.pin,'input_pullup');`

`pinMode(ht.pin);`

which use ESP32 specific code. 

So please test the interval with digitalWrite and pinMode and share the results.

MaBecker, 2019-08-16:
run test by test and share your results

```
// 1. test
setInterval(function() {
        digitalWrite(<your pin>, 0);
        console.log(process.memory());
});}, 1000);
```

```
// 2. test
setInterval(function() {
        pinMode(<your pin>, "output");
        console.log(process.memory());
});}, 1000);
```

```
// 3. test
setInterval(function() {
        pinMode(<your pin>, "input_pullup");
        console.log(process.memory());
});}, 1000);
```

```
// 4. test
setInterval(function() {
        pinMode(<your pin>);
        console.log(process.memory());
});}, 1000);
```

ApocalypseLinyx, 2019-08-16:
Test 1:
```
{ "free": 4038, "usage": 62, "total": 4100, "history": 23,
  "gc": 0, "gctime": 3.158 }
```

Test 2:
```
{ "free": 4038, "usage": 62, "total": 4100, "history": 22,
  "gc": 0, "gctime": 3.158 }
```

Test 3:
```
{ "free": 4038, "usage": 62, "total": 4100, "history": 23,
  "gc": 0, "gctime": 3.151 }
```

Test 4:
```
{ "free": 4039, "usage": 61, "total": 4100, "history": 21,
  "gc": 0, "gctime": 3.144 }
```

repeated without changes

ApocalypseLinyx, 2019-08-16:
If me use read() function once: rh = -1, err = true, checksumError = false every time

if call two read()
```
this.dht.read(function (a) {});
this.dht.read(function (a) {});
```

next call read() get normal rh

memory leak in d virable

```
this.watch = setWatch(function(t) {
    d+=0|(t.time-t.lastTime>0.00005);
  }, ht.pin, {edge:'falling',repeat:true} );
```

For example:
```
d = 10000001001111001000000001101000001001011111000000101000010100000000110100000101011111100000010100100110000000011010000011001011110000001010011001000000001101000001101011110000001010011001000000001100111101101010
010000001010011001000000001100111101101010 ...
```

if add conditional to setWatch function:

```
  this.watch = setWatch(function(t) {
    if(d.length > 41){
      d = d.slice(1)
      d+=0|(t.time-t.lastTime>0.00005);
    } else {
      d+=0|(t.time-t.lastTime>0.00005);
    }
  }, ht.pin, {edge:'falling',repeat:true} );
```

me get normal rh without memory leak.

I need a temp fix, is this true or is there a mistake?

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v04 (c) 2019 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>
{
  "raw": "010000001001110110000000001101011001001110",
  "rh": 63, "temp": 21.4 }
{ "free": 3729, "usage": 371, "total": 4100, "history": 162,
  "gc": 0, "gctime": 3.253 }
_____
{
  "raw": "010000001001110101000000001101011101001110",
  "rh": 62.9, "temp": 21.5 }
{ "free": 3729, "usage": 371, "total": 4100, "history": 162,
  "gc": 0, "gctime": 3.243 }
_____
{
  "raw": "010000001001101110000000001101011001000110",
  "rh": 62.2, "temp": 21.4 }
{ "free": 3729, "usage": 371, "total": 4100, "history": 162,
  "gc": 0, "gctime": 3.249 }
_____
[... 288 lines cut ...]
{
  "raw": "010000001001110110000000001101011001001110",
  "rh": 63, "temp": 21.4 }
{ "free": 3729, "usage": 371, "total": 4100, "history": 162,
  "gc": 0, "gctime": 3.245 }
_____
{
  "raw": "010000001001110100000000001101011001001100",
  "rh": 62.8, "temp": 21.4 }
{ "free": 3729, "usage": 371, "total": 4100, "history": 162,
  "gc": 0, "gctime": 3.24 }
_____
{
  "raw": "010000001001110100000000001101011001001100",
  "rh": 62.8, "temp": 21.4 }
{ "free": 3729, "usage": 371, "total": 4100, "history": 162,
  "gc": 0, "gctime": 3.25 }
_____
{
  "raw": "010000001001110011000000001101010101001010",
  "rh": 62.7, "temp": 21.3 }
{ "free": 3729, "usage": 371, "total": 4100, "history": 162,
  "gc": 0, "gctime": 3.24 }
_____
```

maze1980, 2019-12-22:
Not related to ESP32, it's the script has a bug.
http://forum.espruino.com/conversations/337313/

closed 2020-01-06 by gfwilliams