espruino/Espruino #1689: Building ESP8266_4MB in Docker fails

ipepe, 2019-08-21:
I'm trying to build ESP8266_4MB in Docker but I fail with error:
```
espruino-builder_1  | Installing dev tools for all boards
espruino-builder_1  | Provision BOARDNAME = ALL
espruino-builder_1  | Provision FAMILY =
espruino-builder_1  | ===== ESP32
espruino-builder_1  | pip 19.1.1 from /usr/local/lib/python3.7/site-packages/pip (python 3.7)
espruino-builder_1  | python/pip installed
espruino-builder_1  | pyserial installed
espruino-builder_1  | installing xtensa-esp32-elf-gcc
espruino-builder_1  | Folder found
espruino-builder_1  | GCC is /espruino/xtensa-esp32-elf/bin//xtensa-esp32-elf-gcc
espruino-builder_1  | ===== ESP8266
espruino-builder_1  | installing xtensa-lx106-elf-gcc
espruino-builder_1  | Folder found
espruino-builder_1  | GCC is /espruino/xtensa-lx106-elf/bin//xtensa-lx106-elf-gcc
espruino-builder_1  | ===== LINUX
espruino-builder_1  | ===== RASPBERRYPI
espruino-builder_1  | ===== NRF52
espruino-builder_1  | ===== ARM
espruino-builder_1  | installing gcc-arm-embedded
espruino-builder_1  | Folder found
espruino-builder_1  | Generating pin info
espruino-builder_1  | Generating platform configs
espruino-builder_1  | Generating JS wrappers
espruino-builder_1  | WRAPPERSOURCES = src/jswrap_array.c src/jswrap_arraybuffer.c src/jswrap_dataview.c src/jswrap_date.c src/jswrap_error.c src/jswrap_espruino.c src/jswrap_flash.c src/jswrap_functions.c src/jswrap_interactive.c src/jswrap_io.c src/jswrap_json.c src/jswrap_modules.c src/jswrap_pin.c src/jswrap_number.c src/jswrap_object.c src/jswrap_onewire.c src/jswrap_pipe.c src/jswrap_process.c src/jswrap_promise.c src/jswrap_regexp.c src/jswrap_serial.c src/jswrap_storage.c src/jswrap_spi_i2c.c src/jswrap_stream.c src/jswrap_string.c src/jswrap_waveform.c libs/compression/jswrap_heatshrink.c libs/math/jswrap_math.c libs/graphics/jswrap_graphics.c libs/network/jswrap_net.c libs/network/http/jswrap_http.c libs/network/js/jswrap_jsnetwork.c libs/network/jswrap_wifi.c libs/network/esp8266/jswrap_esp8266_network.c targets/esp8266/jswrap_esp8266.c targets/esp8266/jswrap_nodemcu.c libs/network/telnet/jswrap_telnet.c libs/crypto/jswrap_crypto.c libs/neopixel/jswrap_neopixel.c
espruino-builder_1  | Generating pin info
espruino-builder_1  | DEFINES = -DGIT_COMMIT=552fb8fc -DNO_ASSERT -DRELEASE -DBUILDNUMBER="8" -DUSE_TAB_COMPLETE -DUSE_HEATSHRINK -DUSE_MATH -DUSE_GRAPHICS -DUSE_NET -DUSE_ESP8266 -DUSE_TELNET -DUSE_CRYPTO -DUSE_NEOPIXEL -DUSE_OPTIMIZE_PRINTF -D__ETS__ -DICACHE_FLASH -DXTENSA -DUSE_US_TIMER -DESP8266 -DEMBEDDED
espruino-builder_1  | CC src/jslex.o
espruino-builder_1  | CC src/jsflags.o
espruino-builder_1  | CC src/jsflash.o
espruino-builder_1  | CC src/jsvar.o
espruino-builder_1  | CC src/jsvariterator.o
espruino-builder_1  | CC src/jsutils.o
espruino-builder_1  | CC src/jsnative.o
espruino-builder_1  | CC src/jsparse.o
espruino-builder_1  | CC src/jspin.o
espruino-builder_1  | CC src/jsinteractive.o
espruino-builder_1  | CC src/jsdevices.o
espruino-builder_1  | CC src/jstimer.o
espruino-builder_1  | CC src/jsi2c.o
espruino-builder_1  | CC src/jsserial.o
espruino-builder_1  | CC src/jsspi.o
espruino-builder_1  | CC src/jshardware_common.o
espruino-builder_1  | CC gen/jswrapper.o
espruino-builder_1  | CC libs/compression/heatshrink/heatshrink_encoder.o
espruino-builder_1  | CC libs/compression/heatshrink/heatshrink_decoder.o
espruino-builder_1  | CC libs/compression/compress_heatshrink.o
espruino-builder_1  | CC libs/graphics/bitmap_font_4x6.o
espruino-builder_1  | CC libs/graphics/graphics.o
espruino-builder_1  | CC libs/graphics/lcd_arraybuffer.o
espruino-builder_1  | CC libs/graphics/lcd_js.o
espruino-builder_1  | CC libs/network/network.o
espruino-builder_1  | CC libs/network/socketserver.o
espruino-builder_1  | CC libs/network/socketerrors.o
espruino-builder_1  | CC libs/network/js/network_js.o
espruino-builder_1  | CC libs/network/esp8266/network_esp8266.o
espruino-builder_1  | CC libs/network/esp8266/pktbuf.o
espruino-builder_1  | CC libs/network/esp8266/ota.o
espruino-builder_1  | CC libs/crypto/mbedtls/library/sha1.o
espruino-builder_1  | CC targets/esp8266/uart.o
espruino-builder_1  | CC targets/esp8266/spi.o
espruino-builder_1  | CC targets/esp8266/user_main.o
espruino-builder_1  | CC targets/esp8266/log.o
espruino-builder_1  | CC targets/esp8266/jshardware.o
espruino-builder_1  | CC targets/esp8266/i2c_master.o
espruino-builder_1  | CC targets/esp8266/esp8266_board_utils.o
espruino-builder_1  | CC gen/jspininfo.o
espruino-builder_1  | CC src/jswrap_array.o
espruino-builder_1  | CC src/jswrap_arraybuffer.o
[... 84 lines cut ...]
espruino-builder_1  | ===== RASPBERRYPI
espruino-builder_1  | ===== NRF52
espruino-builder_1  | ===== ARM
espruino-builder_1  | installing gcc-arm-embedded
espruino-builder_1  | Folder found
espruino-builder_1  | BFD: espruino_esp8266_user1.elf: warning: Empty loadable segment detected, is this intentional ?
espruino-builder_1  | 
espruino-builder_1  | BFD: espruino_esp8266_user1.elf: warning: Empty loadable segment detected, is this intentional ?
espruino-builder_1  | 
espruino-builder_1  | BFD: espruino_esp8266_user1.elf: warning: Empty loadable segment detected, is this intentional ?
espruino-builder_1  | 
espruino-builder_1  | BFD: espruino_esp8266_user1.elf: warning: Empty loadable segment detected, is this intentional ?
espruino-builder_1  | 
espruino-builder_1  |   4 -rwxr-xr-x 1 root root   2188 Aug 21 00:11 eagle.app.v6.data.bin
espruino-builder_1  | 472 -rwxr-xr-x 1 root root 482484 Aug 21 00:11 eagle.app.v6.irom0text.bin
espruino-builder_1  |   8 -rwxr-xr-x 1 root root   7540 Aug 21 00:11 eagle.app.v6.rodata.bin
espruino-builder_1  |  28 -rwxr-xr-x 1 root root  28580 Aug 21 00:11 eagle.app.v6.text.bin
espruino-builder_1  |   File "/espruino/ESP8266_NONOS_SDK-2.2.1/tools/gen_appbin.py", line 55
espruino-builder_1  |     print 'file_name cannot be none\n'
espruino-builder_1  |                                      ^
espruino-builder_1  | SyntaxError: Missing parentheses in call to 'print'. Did you mean print('file_name cannot be none\n')?
espruino-builder_1  | make/targets/ESP8266.make:75: recipe for target 'espruino_esp8266_user1.bin' failed
espruino-builder_1  | make: *** [espruino_esp8266_user1.bin] Error 1
espruino-firmware_espruino-builder_1 exited with code 2
```

ipepe, 2019-08-21:
For the record, building ESP32 works fine.

ipepe, 2019-08-21:
I don't know what changed but now it failed at:
```
espruino-builder_1  | SyntaxError: Missing parentheses in call to 'print'. Did you mean print('file_name cannot be none\n')?
espruino-builder_1  | make: *** [make/targets/ESP8266.make:81: espruino_esp8266_user1.bin] Error 1
```

AkosLukacs, 2019-08-22:
Looks like the ESP8266 SDK contains a python file that needs python 2:
 File "/espruino/ESP8266_NONOS_SDK-2.2.1/tools/gen_appbin.py", line 55:
`print 'file_name cannot be none\n'`

The docker file installs python 3. No idea why does it work that way.

MaBecker, 2019-08-22:
Sorry nothing to add, never used that docker file, got the SDK and the gcc local.

gfwilliams, 2019-08-27:
Please could you try editing `Dockerfile` where it says `apt-get install ... `python2.7` to the end?

ipepe, 2019-10-20:
I found a while today, and had success building. I added few changes, so that building in docker is more user friendly. I will submit a PR with changes.

closed 2019-10-20 by ipepe