espruino/Espruino #1691: Accessing `undefined` argument via arguments causes error

gfwilliams, 2019-08-27:
Mentioned here http://forum.espruino.com/conversations/337375

`arguments[1]` causes an error inside the function (a bug), but accessing outside doesn't 

```
>function foo(x,y) {
:  z=arguments;trace(arguments); 
:  for (var i = 0; i < arguments.length; i++)
:    console.log(arguments[i]);   
:}
=function (x,y) { ... }
>foo(1,2) 
#92[r0,l1] Array(2) [ 
  #134[r1,l2] Name Integer 0= int 1

  #121[r1,l2] Name Integer 1= int 2

]
1
2
=undefined
>foo(42) 
#43[r0,l1] Array(2) [ 
  #90[r1,l2] Name Integer 0= int 42

  #136[r1,l2] Name Integer 1    undefined 
]
42
Uncaught ReferenceError: "1" is not defined
 at line 3 col 28
    console.log(arguments[i]);
                           ^
in function "foo" called from line 1 col 7
foo(42)
      ^
>console.log(z[42])
undefined
```

kiranshila, 2020-08-17:
Any movement on this - it is still an issue.

kiranshila, 2020-08-17:
This causes errors all over place for ClojureScript.

```js
(function(){return [x=arguments,x][1][0];})(undefined)
=undefined
```
A neat hack to get around the problem

gfwilliams, 2020-08-18:
Just fixed

closed 2020-08-18 by gfwilliams

kiranshila, 2020-08-18:
You're a machine @gfwilliams! Thanks!

commit 7a938ce0 (2020-08-18): Fix ReferenceError when arguments[0] is used when an argument is undefined (fix #1691)