espruino/Espruino #1693: Add I2S support for ESP8266

Piruzzolo, 2019-09-07:
It would be great!

gfwilliams, 2019-09-09:
What do you actually want the support for? As far as I remember the `neopixel` library uses it under the hood, but trying to actually do audio could be problematic with Espruino's low processing speed.

Piruzzolo, 2019-09-09:
To get sound samples from a i2s mic.

To make it fast, it should call a device-specific C function, I guess. Then offer a common API to access i2s

MaBecker, 2019-09-11:
> To get sound samples from a i2s mic.
> 
> To make it fast, it should call a device-specific C function, I guess. Then offer a common API to access i2s

Are you planing to implement i2s as c-code?

MaBecker, 2020-03-13:
> @MaBecker I dubt I could implement it properly, if I tried. But maybe someone has already done the low level work. Need to google a bit

Well, try and find out :)

MaBecker, 2020-09-24:
Look's like no one likes to jump in to add this and there haven't been  further request in the forum.

closed 2020-09-24 by MaBecker