espruino/Espruino #1698: Support Wemos D1 Mini Pro (v1.1.0) - ESP8266 with 16MB?

ipepe, 2019-10-20:
Hey.

Today I had success building my own image of ESP8266 for use with SDCard. I wanted to ask for help regarding image for 16MB flash in ESP8266. I noticed that there are 2 build methods for building Espruino for ESP8266 - one for 512KB (if I'm not mistaken?) flash memory and 4MB. I have one that has 16MB and feel like there are some options that would allow me to use more Flash memory, right? Probably higher number of variables? Could You help me with all these parameters and stuff?

Although I understand it might be problematic, as I found following information:
https://github.com/micropython/micropython/issues/2335

My Wemos D1 Mini Pro:
https://wiki.wemos.cc/products:retired:d1_mini_pro_v1.1.0

MaBecker, 2019-10-20:
Well, 16MB is just 12MB more flash space that can be used to store objects, it does not mean that there are more vars possible.

Only one MB is memory mapped, the one that booted from. All other segments need flash write and read. Which means they needed to be read into ram to be used.

This is the function where to add the extra flash areas:

https://github.com/espruino/Espruino/blob/11a9bd79c40357b77f9b8ef7d61b21bcfd3a7b5b/targets/esp8266/jshardware.c#L1406

Next is to use 16MB special flash parameter, check the esptool readme

https://github.com/espressif/esptool/blob/master/README.md

MaBecker, 2020-02-05:
@ipepe any progress?

closed 2020-03-13 by MaBecker