espruino/Espruino #1701: Bangle.js hid and uart conflict

jasnell, 2019-11-12:
When using `NRF.setServices()` and setting both `uart:true` and `hid` to a hid report, the WebIDE will not properly connect. The WebIDE will see the bangle and will attempt connection but it will hang.

gfwilliams, 2019-11-12:
This on Windows 10 again?

I can connect fine with Chrome on my phone (and nRF Connect), and with the Noble-based command-line app. But I can't connect with Chrome on Linux either.

I think this is a Web Bluetooth issue :(

jasnell, 2019-11-14:
Yeah, this is that Windows 10 issue. Investigating further it definitely looks to be specific to web bluetooth on Windows so likely just something that needs to be documented as a known issue. I've tried a number of things to work around it and nothing is working.