espruino/Espruino #1702: Enhancement for  Graphics

MaBecker, 2019-11-12:
This is just a collection of function that came in my mind when working with the excellent library.

For a better readability, will try to write one comment for each enhancement

MaBecker, 2019-11-12:
* extend circle to draw arcs

MaBecker, 2019-11-12:
* extend line with thickness

MaBecker, 2019-11-12:
* extend draw  with rotation possibility

MaBecker, 2019-11-12:
* where to find details for this nice feature used in Bangle.js app "GPS Time"

```
setInterval(function() {
  g.drawImage(img,48,48,{scale:1.5,rotate:Math.sin(getTime()*2)/2});
},100);

```

this would nice to have for poly functions.

gfwilliams, 2019-11-12:
https://banglejs.com/reference#l_Graphics_drawImage

Those definitely seem like good things to add.

I was considering maybe moving to a better fillPoly (that could do irregular polys) then handling drawLine using that. The real bonus of a line width is you could use the HersheyText line fonts, which scale *really* nicely

MaBecker, 2019-11-14:
At the moment I use this workaround, which allows to scale and translate a array with pairs of points..

```
function translate(x, y, p) {
    p.forEach((e, i) => {
        p[i] += (i % 2) ? y : x;
    });
    return p;
}

function scale(x, y, p) {
    p.forEach((e, i) => {
        p[i] *= (i % 2) ? y : x;
    });
    return p;
}

var plStop = [0,0,20,0,20,20,0,20];
g.fillPoly(translate(40,170,scale(2,2,plStop)));
var plPlay = [0,0,20,10,0,20];
g.fillPoly(translate(160,170,scale(2,2,plPlay)));
```

MaBecker, 2019-11-18:
* process bar, slider .....

gfwilliams, 2019-11-19:
allObjects has a UI library that looks really promising which just needs adding to the EspruinoDocs repo with all his documentation - I just haven't had the time so far

MaBecker, 2019-11-19:
Yes this UI is very cool 😎

MaBecker, 2019-12-07:
####  Replace optimized Ellipse with standard Bresenham algorithm to remove the spikes.

Eye draw test on Bangle.js

Source

```
g.clear();
g.setColor(1,1,1).fillEllipse(40,60,120+80,180);
g.setColor(0,0,1).fillCircle(120,120,50);
g.setColor(0,0,0).fillCircle(120,120,20);
g.setColor(1,1,1).fillCircle(112,112,5);
```

Result:

[image]

MaBecker, 2019-12-08:
#### Hershey Fonts

Started with diving into the hershey world.  This is extremely cool.
http://coopertype.org/event/the_hershey_fonts
https://emergent.unpythonic.net/software/hershey
http://paulbourke.net/dataformats/hershey/

And now starting to code in javascript to figure out how simple this can be to  implement into Graphics library.

implement:
* JSGRAPHICS_FONTSIZE_HERSHEY = 0
* JSGRAPHICS_LINETHICKNESS = 1
* drawLine with line thickness
* setFontHershey(size,thickness)
* graphicsDrawHersheyChar()
* ....

gfwilliams, 2019-12-09:
Do you have a PR for the new ellipse? ;)

MaBecker, 2019-12-09:
Yep, just added pr #1720

MaBecker, 2019-12-11:
Would like to implement thickness for drawLine. Can you please advise if and how it should be implemented.

gfwilliams, 2019-12-11:
Sounds like a good plan. I'd store lineThickness the same way we do fontAlign/etc in graphics, then check in drawPoly and run a separate bit of code. It should be ifdef'd with SAVE_ON_FLASH though.

The easy way to do it would be to draw each line segment as a poly as you've done, but I think that's going to be too slow if we want to replace Vector with Hershey fonts (which I guess is the main goal?) as there'd be loads of overdraw, and really we want to work on the whole polyline

The actual integration with graphics would take a while and would be a waste of your time, so if you could come up with an algorithm that would turn a polyline into a filled poly then I could integrate it. Even doing the algorithm in JS would work.

Something line this - you'd need to generate the red points:

[image]

In your code above i notice you're taking the angle and the using sin and cos, but that's very slow and you can actually just use the difference between coordinates:

```
dx = x2-x1;
dy = y2-y1;
d = sqrt(dx*dx + dy*dy);
dx = dx*lineWidth/d;
dy = dy*lineWidth/d;
// then [dy,-dx] is 90 degrees, [-dy,dx] is the other way
// [dx,dy] is in-line, so [(dx+dy)/2, (dy-dx)/2] 
```

You can even compare dx/dy with the *next* dx/dy using a cross product in order to work out which way the line bends (and how much) to figure out if points need adding or removing.

gfwilliams, 2019-12-11:
... only saw your update with the Hershey font you printed while writing this. It looks great - it's hard to see how anyone would complain if the font changed :)

MaBecker, 2019-12-11:
Great, thanks for you guiding informations.

MaBecker, 2019-12-12:
Just a short update on testing hershey font using a [C-Style Coordinate array for the SIMPLEX character set ](http://paulbourke.net/dataformats/hershey/) instead of decoding [jhf](https://github.com/kamalmostafa/hershey-fonts/blob/master/hershey-fonts/rowmans.jhf).

Mirrored and move chars to have same height and baseline. Also adjusted some points for a nice shape.

Started with numbers 0 - 9, scaling 1.5

[image]

```
/*
 var hfl =  [
        width, heigth, numLines,
        [ fist line x,y ],
        ...
        [ last line x,y ]
    ];
*/
  ......
   // /* 1 Ascii 49 */
    [
        6, 22, 1,
        [0, 5, 2, 3, 5, 0, 5, 21]

    ],
```  

function to setLineWith

```
function setLineWidth(x1, y1, x2, y2, lw) {
    var dx = x2 - x1;
    var dy = y2 - y1;
    var d = Math.sqrt(dx * dx + dy * dy);
    dx = dx * lw / d;
    dy = dy * lw / d;

    return [
        // rounding
        x1 - (dx + dy) / 2, y1 - (dy - dx) / 2,
        x1 - dx, y1 -dy,
        x1 + (dy - dx) / 2, y1 - (dx + dy) / 2,

        x1 + dy, y1 - dx,
        x2 + dy, y2 - dx,

        // rounding
        x2 + (dx + dy) / 2, y2 + (dy - dx) / 2, 
        x2 + dx, y2 + dy,
        x2 - (dy - dx) / 2, y2 + (dx + dy) / 2, 

        x2 - dy, y2 + dx,
        x1 - dy, y1 + dx
    ];
}

g.fillPoly(setLineWidth(20,20,100,200,5));

```

[image]

Not sure about those six rounding lines.

MaBecker, 2019-12-12:
First result with adding line width to poly segments

[image]

Fontsize: 66px

MaBecker, 2019-12-12:
working with scaling, thickness and kerning every one can create his individual look

hf = { sx: 2.5, sy :5,  t: 5, k: 5 };
sx : font width 
sy : font height, default 22px
t: thickness in px
k : kerning in px

[image]

MaBecker, 2019-12-12:
Or as small font with  hf = { sx: 1, sy :1, t: 1, k: 2 };

[image]

MaBecker, 2019-12-13:
Maybe add a function like this to handle the style. 

     g.setFontHershey({width: in_px, heigth : in_px, bold : in_px, kerning : in_px});

MaBecker, 2019-12-13:
Before starting to convert hershey font files to header file, some advise for output format would be helpful to create eg hershey_romans_font.h

    static const unsigned char hersheyFontPolys[] IN_FLASH_MEMORY = {
     // Character code 32
    ...
     // Character code 255
     }

gfwilliams, 2019-12-13:
Nice! About the rounding, try changing `x1 - (dx + dy) / 2, y1 - (dy - dx) / 2,` to `x1 - (dx + dy) * 0.71, y1 - (dy - dx) * 0.71,` - it should make it more rounded :)

gfwilliams, 2019-12-13:
In other fonts I've had one array for character data, one array for # of points. It means you have to iterate through to find each character, but it immediately shaves 100 bytes off the structure because you can use 8 bits rather than 16 for the length.

MaBecker, 2019-12-13:
> In other fonts I've had one array for character data, one array for # of points

Yes, seem to be the easiest way to implement it, like you did for the vector font.

MaBecker, 2019-12-16:
Got a version javascript version running 

https://gist.github.com/MaBecker/89a8ec3314f456ccfe397593783ec8b4

What about spacing between chars?

MaBecker, 2019-12-16:
Decided to use different factor for drawing the rounded edges.

[image]

Is there a solution to handle yellow marked section, because of the slope the width is to small

gfwilliams, 2019-12-17:
Are you actually doing what's suggested in https://github.com/espruino/Espruino/issues/1702#issuecomment-564541847 ? Looks like you might just be producing one polygon per line segment?

Doing one big poly would go a long way towards fixing those glitches, but I had considered maybe making the internal poly fill algorithm take coordinates that were 16x bigger (so 16 = 1 pixel, 32 = 2, etc) would help with issues like this

MaBecker, 2019-12-17:
> Looks like you might just be producing one polygon per line segment?

Yes, just started working on this :)

> making the internal poly fill algorithm take coordinates that were 16x bigger (so 16 = 1 pixel, 32 = 2, etc) would help with issues like this

Good point, will include it.

MaBecker, 2020-02-12:
Just added this pr https://github.com/espruino/Espruino/pull/1757 for g.quadraticBezier()

MaBecker, 2020-03-13:
* [x] draw rectangle with rounded corners

http://forum.espruino.com/conversations/344773/

Would be easy to add to firmware.

gfwilliams, 2020-05-13:
Here's my attempt at Hershey rendering - creating one big poly rather than one per line:

https://www.espruino.com/ide/emulator.html?gist=663f7d6e280e33511881a3dfff2ee30f&upload

Still a few edge cases, but reasonably quick and significantly less overdraw.

[image]

MaBecker, 2020-05-13:
Wow, just tested, very nice!

Have you thought about about chars > 127 and < 256 ?

gfwilliams, 2020-05-13:
It'll be much quicker when it's converted to C too :)

Potentially we could add them, yes. Euros and degrees at least.

One of the reasons for doing this was to reduce flash usage though so I don't want to bump it up too much!

I'm not 100% on the characters but it strikes me that for many of them we could maybe hardcode it? So Ü = U + umlaut ?

gfwilliams, 2020-05-13:
Obviously there's still the fillPoly issue. You wouldn't think it'd be so hard to get a reliable polygon fill algorithm!

MaBecker, 2020-05-13:
> I'm not 100% on the characters but it strikes me that for many of them we could maybe hardcode it? So Ü = U + umlaut ?

Yes they are called Umlaute  äöüÄÖÜ plus €ß° 

ü = u + umlaut
Ü = U + same as umlaut, just on a different position
€: parts from O plus two lines
ß: parts of 3 plus a line 

I was spending many hours scanning other graphic libs for a smart, short,  quick and working  algorithm, without success. fillPoly is not as simple as it looks.

MaBecker, 2020-05-13:
I like the look and feel of the chars with debug flag!

gfwilliams, 2020-05-18:
Ok, change of plan again. A custom-made vector font seems preferable: https://github.com/espruino/Espruino/issues/1824

MaBecker, 2020-07-10:
time to close this ;-)

closed 2020-07-10 by MaBecker