espruino/Espruino #1703: Locale for date and time

MaBecker, 2019-11-20:
What about  date and time in locale specific formats? 

Not as part of firmware but as part of user javascript code.

* Time: 24 hour format or 12hour with am and pm

```
var time = new Date();
console.log(
  time.toLocaleString('en-US', { hour: 'numeric', hour12: true })
); 
```

* Date: order of  day and month
* Date: name of days and month

https://github.com/espruino/Espruino/blob/449b1f8db57486f389133224f4dced220ab9b78d/src/jswrap_date.c#L29-L30

Maybe assign lookup array for day and month names from customer code to a time format translation function in the firmware. (add prototype function to Date).

gfwilliams, 2019-11-21:
I'm not a massive fan of sticking it inside Espruino itself, because it's a lot of extra bulk when templated strings let you do format date reasonably easily.

It's kind of annoying that [Date.toLocaleFormat](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Date/toLocaleFormat) isn't a standard, as I'd be totally behind putting something like that in.

Looking at the MDN docs there's a standard for `Intl` (https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DateTimeFormat) that I reckon could be implemented as a JS library reasonably easily?

MaBecker, 2019-11-21:
Completely agreed, do not blow up the firmware with locale.

Will check for tiny js lib. There is a nice lib from @allObjects in the forum, will check this too.

MaBecker, 2019-11-21:
From the forum: [Date().toDateString()](http://forum.espruino.com/conversations/290058/)

MaBecker, 2020-03-13:
Ok, locale is now available as Bangle.js app.

How could this be available for all other devices as well?

* Implement a locale uploader into the WebIDE?
or
* use Espruino comandline tool to upload locale?
or
* ?

gfwilliams, 2020-03-13:
Maybe make this same code use UART.js to allow an upload to *any* Espruino device? https://github.com/espruino/EspruinoWebTools/blob/master/uart.js

We could add it to the Espruino site in a similar way to the Image Converter

MaBecker, 2020-03-13:
> We could add it to the Espruino site in a similar way to the Image Converter

 I like that idea, pick a language and get something like this: 

```
require('Storage').write('local', ` .........`);
```

gfwilliams, 2020-03-13:
Yes - it could even write it automatically if needed.

gfwilliams, 2020-07-14:
Closing this - now implemented for Bangle.js

closed 2020-07-14 by gfwilliams

MaBecker, 2020-07-14:
Well, would be nice to  have locale for all other devices too.

MaBecker, 2020-07-14:
Maybe as function in the WebIDE

gfwilliams, 2020-07-14:
Yes, that sounds like a good idea.