espruino/Espruino #1704: Bangle.js enhancement requests

MaBecker, 2019-11-21:
Maybe this could be used to list/add nice Bangle.js enhancement requests.

MaBecker, 2019-11-21:
Bangle.on("swipe") to identify swipes  

* BTN4 to BTN5 : right swipe 
* BTN5 to BTN4 : left swipe

gfwilliams, 2019-11-21:
> Bangle.on("swipe")

I was working on this yesterday ;)

MaBecker, 2019-11-22:
> > Bangle.on("swipe")
> 
> I was working on this yesterday ;)

Just tested, works great!

MaBecker, 2019-11-22:
Have you thought about a long button press that calls a inc() or dec() function.
Want to change a value from 50 to 51, easy one press, What about a change to 75. that need many presses.

Touch once, tigger a interval to call a inc or dec function, release clear intervall and you are good.

Maybe like this, done in javascript  and input from allObjects

[button_long_press_inc_dec](https://www.dropbox.com/s/7wc44u6j106k8qs/button_long_press_inc_dec_v2.m4v?dl=0)

gfwilliams, 2019-11-25:
You mean in `E.showMenu` when changing a menu item? Sounds like a good plan. The menu is actually all done in JS anyway so it's not too hard to add: https://github.com/espruino/Espruino/tree/master/libs/js/banglejs

MaBecker, 2019-11-26:
> You mean in `E.showMenu` when changing a menu item? Sounds like a good plan. The menu is actually all done in JS anyway so it's not too hard to add: https://github.com/espruino/Espruino/tree/master/libs/js/banglejs

For the menu it would be nice to be able to scroll in loop, will have a look into it.

> Touch once, tigger a interval to call a inc or dec function, release clear intervall and you are good.

https://gist.github.com/MaBecker/fd482543d0b5edf0692baab2a9f2e293

Keep it as JS or add to Bangle firmware?

gfwilliams, 2019-11-27:
If you want to add it to the firmware then you could add it to the menu function (as JS) here: https://github.com/espruino/Espruino/blob/master/libs/js/banglejs/E_showMenu.js

Although right now it uses just the 3 buttons, with the middle button toggling between editing and not (which IMO makes sense if you're wearing gloves). Honestly, the whole built-in E.showMenu needs some love on Bangle.js (eg. colour for highlights, more obvious icons).

I'm not sure this is worth adding to the firmware as a 'general purpose' function though as figuring out the API would probably be as hard as writing a setInterval yourself.

gfwilliams, 2020-03-05:
Closing this for now - there have been loads of showMenu changes & tweaks. Auto-stepping isn't in there, but rolling over and highlights are.

If more people complain about auto-stepping maybe we could add it, since I've just had to take graphical_menu out and do a custom one for Bangle.js (to do translations)

closed 2020-03-05 by gfwilliams