espruino/Espruino #1710: Should g.fillPoly handle a complex path?

MaBecker, 2019-12-01:
got some drawPoly that are not rendered like I though it should.

```
var flake1 = [114,120,126,120,126,150,150,168,144,180,126,168,126,192,114,192,114,168,96,180,90,168,114,150,114,120];
var flake2 = [114,20,126,20,126,50,150,68,144,80,126,68,126,92,114,92,114,68,96,80,90,68,114,50,114,20];

g.clear();
g.drawPoly(flake1);
g.fillPoly(flake2);
```

[image]

gfwilliams, 2019-12-02:
Right now this is as expected - it's what I'd mentioned before about having a slower but more capable fillPoly implementation. The current one works on a scanline basis - filling from lowest X to highest X.

MaBecker, 2019-12-05:
> It's something I plan to improve at some point soon though - it'll likely decrease the rendering speed slightly but it definitely seems like a worthwhile addition.

Taken from the forums post

gfwilliams, 2019-12-09:
Example algorithm posted here: http://forum.espruino.com/conversations/341492/#15015802

It looks simpler than what's there already really!

gfwilliams, 2019-12-09:
It's simple because it doesn't work properly :/

Current code is here: https://github.com/espruino/Espruino/compare/fillPoly_irregular

I had to modify it so it handled the topmost Y coordinate correctly (eg when drawing a diamond it'd miss off the top), and then also had to add a 'slope' so that it could handle the duplicate corners that would then be produced.

However drawing vector fonts still results in something like this:

```
................................
................................
................................
..#####...............#####.....
..#####..............######.....
...#####............######......
....#####..........######.......
....######........#######.......
.....######.......######........
......#####......######.........
......######....######..........
.......######..#######..........
........#############...........
.........######.................
.........###########............
..........#########.............
...........#######..............
...........######...............
...........######...............
..........########..............
.........######.................
.........##########.............
........######..####............
.......######...#####...........
......######.....#####..........
......######......####..........
.....######.......#####.........
....######.........#####........
...######...........#####.......
...######...........#####.......
..######.............#####......
.######...............#####.....

```

This is because the vector font is (currently) made out of a bunch of regular polygons, and there appear to be issues where they join

gfwilliams, 2019-12-09:
Ok, now fixed. Looks like the code mentioned initially actually does a standard polygon fill thing, which is to not fill the top and right hand pixels so that if you put polygons against each other you don't end up with overdraw - which is why the scanline code refused to draw the top pixel

My hacks will likely cause problems with more complex polygons since i'm effectively duplicating vertices at the corners :(

MaBecker, 2019-12-09:
[image]

Nice, many thanks

gfwilliams, 2019-12-10:
Ok, fixed with https://github.com/espruino/Espruino/commit/4442951586b1347045e0ceafc8da6ffc93c88ac6

closed 2019-12-10 by gfwilliams