espruino/Espruino #1722: parseInt("0b",16)

gfwilliams, 2019-12-16:
Not sure how this one slipped through, but it looks like even if you explicitly set the radix, Espruino can be tripped up:

```
// espruino
>parseInt("0b",16)
= 0
// node.js
> parseInt("0b",16)
= 11
```

MaBecker, 2019-12-16:
Just tested, only wrong for "0b" and "0B"

"0a,0c-0f" and "0A,0C-0F"  are correct

MaBecker, 2019-12-16:
>parseInt("0x0b",16);
=11

https://github.com/espruino/Espruino/blob/3251c043d6afad2e44f219b6c003ed4555faa4c9/src/jsutils.c#L122-L128

gfwilliams, 2020-01-28:
Wow, this is a mess: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/parseInt

Sometimes I hate JS:

```
parseInt("0b01",2)
0
parseInt("0b01")
0
Number("0b01")
1
> parseInt("066")
66
> Number("066")
66
> 066
54
```

So `0x...` is actually a special-case for parseInt

commit 57da2618 (2020-01-28): Fix `parseInt("0b",16)` as well as some other non-compliant behaviour (fix #1722)
closed 2020-01-28 by gfwilliams