espruino/Espruino #1728: ESP32 support for 16-byte characteristic UUIDs

mfikes, 2019-12-17:
If you use Espruino master to set up a BLE service

```javascript
NRF.setServices({
  "6fe0fede-40e7-45e6-a91f-3422d2dbf6ed" : {
   "236e054d-d196-4b0c-b984-b7e6846b2ba7" : {
      value : "Hello",
      readable : true
    }
  }
});
```

you will see that the characteristic is not placed under the full 16-byte UUID. For the example above, it appears to be advertised as 0x054D.

My hunch is that, if the BLE characteristic [initialization code](https://github.com/espruino/Espruino/blob/3251c043d6afad2e44f219b6c003ed4555faa4c9/targets/esp32/BLE/esp32_gatts_func.c#L438) currently only supports 2-byte UUIDs.

(On the other hand, service UUIDs empirically appear to be [properly initialized](https://github.com/espruino/Espruino/blob/3251c043d6afad2e44f219b6c003ed4555faa4c9/targets/esp32/BLE/esp32_gatts_func.c#L495) on the ESP32 and can handle 16-byte UUIDs.)

Also, the characteristic initialization issue is fixed, then [this line](https://github.com/espruino/Espruino/blob/65efbe2f69b539587427e7cdf8e9e14ef36f058d/libs/bluetooth/bluetooth_utils.c#L193) could be removed, as it is assigning the 2-byte UUID field for a UUID that is known to be a 16-byte UUID.

MaBecker, 2020-03-13:
@mfikes do you like to create a pull request to make this work?

gfwilliams, 2024-02-02:
Pretty sure I fixed this now

closed 2024-02-02 by gfwilliams