espruino/Espruino #1730: ESP32 NRF.updateServices characteristic / service mixup

mfikes, 2019-12-17:
Following the example in the [NRF.updateServices documentation](https://www.espruino.com/Reference#l_NRF_updateServices), directly calling `updateServices` instead of as a result of a button push:

```
>NRF.setServices({
:  0xBCDE : {
:    0xABCD : {
:      value : "Hello",
:      maxLen : 20,
:      notify: true
:    }
:  }
:});
=undefined
>NRF.updateServices({
:    0xBCDE : {
:      0xABCD : {
:        value : "World!",
:        notify: true
:      }
:    }
:  });
FIXME
Uncaught Error: Unable to find service with UUID 0xabcd
 at line 8 col 4
  });
   ^
```

You can see that you get an error that seems to imply a mixup between the service and characteristic identifier.

mfikes, 2019-12-17:
The [`FIXME` output](https://github.com/espruino/Espruino/blob/dc10d8934117f3e15a7612baaae7afed26b0e9f6/libs/bluetooth/bluetooth_utils.c#L257) is indicating that this is not yet implemented for the ESP322.

alfierivera, 2020-01-03:
Any ideas of how much effort will this take to be implemented? I might be able to sponsor such a feature, as I need it for an experiment I'm making.

gfwilliams, 2020-01-06:
For me personally to do it I'd imagine between 0.5 and 1 day, so £500-1000 ish. Others may be cheaper though.

@mfikes if you're interested feel free to post up what you'd be willing to do it for. I have way too much work on my plate right now so if you're interested please don't have any concerns about stealing work from me :)

MaBecker, 2020-03-13:
@mfikes have you been working on this. If yes please share - Thanks.

mfikes, 2020-03-16:
@MaBecker No, I have not been working on this.

MaBecker, 2020-03-16:
Would you like too?

closed 2024-04-24 by MaBecker