espruino/Espruino #1731: ESP32 app and esp-idf file stucture

MaBecker, 2019-12-26:
Started compiling Espruino for ESP32 after a long pause.

Wow is is completely different to building ESP8266, got me a little confused.

Completely dislike how Espressif is shipping this, not sure if this is because just started, but spreading libs and include over many folders is a bad style.

Is it possible to mode all libs to a single lib folder and the same for includes?

MaBecker, 2019-12-27:
@wilberforce are still developing for ESP32?

gfwilliams, 2020-01-06:
I haven't looked into it too deeply, but from experience with other manufacturer's HALs it's probably better to leave the manufacturer supplied folders as near standard as they can be - even if you can make it nice once, it'll be a nightmare for the next release

MaBecker, 2020-01-06:
> I haven't looked into it too deeply, but from experience with other manufacturer's HALs it's probably better to leave the manufacturer supplied folders as near standard as they can be - even if you can make it nice once, it'll be a nightmare for the next release

Yes good point, so keep it and get used to it.

closed 2020-01-06 by MaBecker