espruino/Espruino #1733: GPS gives wrong longitudes in case of 25 degrees east

ikkala, 2019-12-27:
Issue detected by using Bangle.js code when longitude is 025 degrees east:

```
Bangle.on('GPS', function(fix) {
    console.log('fix:', fix);
});
Bangle.on('GPS-raw', function(nmea) {
    console.log('nmea:', nmea);
});
```
This produces:

> fix: { "lat": (censored), "lon": **21.73**(censored), "alt": 130.2, "speed": 0.468556,
>   "course": NaN,
>   "time": Date: Fri Dec 27 2019 15:44:45 GMT+0200,
>   "satellites": 5, "fix": 1 }
> nmea: $GNGLL,(censored),N,0**2544**.(censored),E,134445.00,A,A*71

So, raw GPS data is correct but parsed GPS fix is incorrect.

By reading source code one can find `nmea_decode` function line  https://github.com/espruino/Espruino/blob/master/libs/misc/nmea.c#L94 setting longitude by using `nmea_decode_latlon` function which uses helper functions `stringToFloat` and `stringToInt` to parse the NMEA data to longiture (and to latitude). Also, when continuing to the implementation of of `stringToInt` in https://github.com/espruino/Espruino/blob/master/src/jsutils.c#L164 you can find that eventually the `getRadix` incorrectly deduces that the radix of the integer is 8. So, by giving string "025" it will think it's octal and in decimals you get 2*8+5=21.

Like in implementation of `parseInt`, one could monkeypatch this like done in https://github.com/espruino/Espruino/issues/538 -- in NMEA case you can always put radix to 10 explicitly, also for float part of the longitude (and latitude).

Therefore, proposal for fix is:

```
double nmea_decode_latlon(char *nmea, char *comma) {
  if (*nmea==',') return NAN; // no reading
  char *dp = nmea;
  while (*dp && *dp!='.' && *dp!=',') dp++; // find decimal pt
  *comma = 0;
  double minutes = stringToFloatWithRadix(&dp[-2], 10, NULL);
  *comma = ',';
  dp[-2] = 0;
  int x = stringToIntWithRadix(nmea, 10, NULL, NULL);
  return x+(minutes/60);
}
```

I was not yet brave enough to do pull request because building C code is not my profession currently.

This might be related to #1722 too?

ikkala, 2020-01-02:
Managed to do create local build and tested that the fix works. Will create pull request.

gfwilliams, 2020-01-06:
Thanks for this - sorry for the slow replies but this looks spot on.

PR #1735 (2020-01-06): Parse GPS longitudes (and latitudes) in radix 10
Fixes #1733

Before this fix, for example longitude 025 degrees was parsed to 21 because parseInt interprets the longitude as octals.
commit ac888077 (2020-01-06): Parse GPS longitudes (and latitudes) in radix 10.

Fixes #1733
closed 2020-01-06 by gfwilliams