espruino/Espruino #1734: Wio LTE Error on SMS recieve

dimik, 2019-12-28:
Hi, i tried Wio LTE demo code from [here](http://www.espruino.com/WioLTE).
Everything seems to work fine, except receiving SMS code fails.
First SMS it shows the message text and error, then only error.

```javascript
  // Handle SMS coming
  board.on('message', function(id){
    board.SMS.read(id, function(d, sms){
      if(d !== "OK") throw new Error(d);
      console.log('SMS from:', sms.oaddr);
      console.log(':', sms.text);
    });
  });
```

First SMS:

```
] "\r"
] "\n+CMTI: \"ME\",3\r\n"
["AT+CMGR=3\r\n"
] "\r"
] "\n+CMGR: \"REC UNREAD\",\"___number___\",,\"19/12/28,14:05:"
] "40+28\"\r\nHello world \r\n\r\nOK\r\n"
SMS from: ___number___
: Hello world
Uncaught Error: Function "cmd" not found!
 at line 1 col 3
b.cmd("AT+QGPSLOC?\r\n",2E3,e)
  ^
in function called from system
```

Second SMS:
```
] "\r"
] "\n+CMTI: \"ME\",4\r\n"
Uncaught Error: Function "cmd" not found!
 at line 1 col 9
var c;b.cmd("AT+CMGR="+a+"\r\n",2E3,function g(a){if(void 0!...
        ^
in function "read" called from line 58 col 6
    });
     ^
in function called from system
> 
```

gfwilliams, 2020-01-08:
Are you running the full unmodified demo code from http://www.espruino.com/WioLTE#using-lte-and-gps 

Because `AT+QGPSLOC` occurs when you call `geoLocStart` for geolocation so this may not be SMS related at all.

gfwilliams, 2024-02-02:
No answer, and Seeed stopped supporting Wio LTE several years ago

closed 2024-02-02 by gfwilliams