espruino/Espruino #1738: Bangle.js options for when screen turns on

gfwilliams, 2020-01-07:
`Bangle.setOptions` should have options for:

* Whether accelerometer can turn the LCD on
* accelermeter values needed to turn the LCD on
* Whether buttons turn the LCD on

http://forum.espruino.com/conversations/342080/#comment15039100

fanoush, 2020-01-29:
Related idea - some watches have quiet/night mode - time interval when it doesn't turn screen on by movement (and vibrate on messages). Or there could be functionality like switch to quite mode now until next alarm clock rings - when setting alarm in some clock app before going to sleep. The point is - if possible do not hardcode everything in C code to allow flexibility like this.

Hopefully with settings already mentioned in previous post this can be done just by some setTimeout scheduled for specific times and changing those values.

gfwilliams, 2020-02-06:
Just added the setOptions - and yes, I reckon that should be pretty easy to handle a night mode - maybe a widget that changed the settings (beep/buzz/wake on face up/etc) at the correct times

commit 84248c8c (2020-02-06): Bangle.js: add Bangle.setOptions({wakeOn*}) for when LCD should light (fix #1738)
closed 2020-02-06 by gfwilliams