espruino/Espruino #1749: digitalPulse() not working as expected

MaBecker, 2020-01-20:
see http://forum.espruino.com/comments/15067891/ for details.

sleuthware, 2020-01-20:
Sun 2020.01.19

I made a later comment post 9 at:
> http://forum.espruino.com/comments/15067935/

regarding non whole numbers within an array. I was working with the MDBT42Q breakout board and ran into a simmilar issue. While still unresolved and not fully researched at 'C' code level, I seem to run into issues when pulses got down to the 200usec range, approx 8000Hz. The docs really don't explain the limits here, and I still can't locate where the under the hood timing stuff takes place.

Simple math 1 / 8000 = 125usec

As an observation, if the divisor is from a (not chceked) 16MHz clock crystal, this might explain things.

Working on a project/tutorial to replace an Arduino interrupt driven frequency counter example, but need pulses around 1usec width and detection around 10usec duration. Would this be possible?

Will continue to gather test data to determine current limits with 2v04 and when sufficient data has been gathered, start a forum post.

Robin

gfwilliams, 2020-01-20:
I think the issue you're hitting here is that the timings on ESP8266 are most likely done with the 32kHz RTC. Same thing happens on the NRF52, however STM32 based boards do some crazy stuff to allow them to use a much more accurate timer.

While the high speed timer used for digitalPulse is more accurate, it uses the RTC to re-sync itself, and that means that the actual pulse accuracy for long trains of pulses is pretty limited.

Basically this is the last bullet point on https://github.com/espruino/Espruino/issues/1444

We could switch to *not* using the RTC, but that would mean that if for instance you did a 4000Hz waveform capture of 4000 samples, then a `setTimeout` for 1000ms, the two may not get called at quite the same time

closed 2020-10-07 by gfwilliams

gniezen, 2020-11-02:
Why was this change reverted? Should the issue maybe be re-opened?

gfwilliams, 2020-11-03:
I forget where the discussion on this was now... It didn't seem to make things noticeably better, and was a bit too risky right before the 2v08 release. I've just re-applied it though so I'd be interested to see whether you find real improvements with it

gniezen, 2020-12-14:
I compiled and flashed the latest master branch, but it doesn't appear to have made a difference. I'm still seeing pulse trains of 30ms or more, when they should be 27.5ms (referring to http://forum.espruino.com/conversations/354435/).

commit 0c6b0c82 (2020-10-07): Util Timer no longer uses RTC - works based on estimated time from the hardware timer itself (fix #1749, ref #1444)
commit 4ea13400 (2020-10-09): Revert "Util Timer no longer uses RTC - works based on estimated time from the hardware timer itself (fix #1749, ref #1444)"

This reverts commit 0c6b0c8289347163764cb654feb7197fba8d0bf5.
commit fbdb6c70 (2020-11-03): Util Timer no longer uses RTC - works based on estimated time from the hardware timer itself (fix #1749, ref #1444)