espruino/Espruino #1751: Building for ESP32

bazooka07, 2020-02-03:
Hello
I have build myself Espruino.
But when i boot the esp32, I have the following message in the terminal :
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2668
load:0x40078000,len:7304
load:0x40080000,len:5312
entry 0x40080274
E (529) esp_image: Image length 1392896 doesn't fit in partition length 1376256
E (529) boot: Factory app partition is not bootable
E (529) esp_image: image at 0x160000 has invalid magic byte
E (535) boot: OTA app partition slot 0 is not bootable
E (540) boot: No bootable app partitions in the partition table
Fatal exception (0): IllegalInstruction
epc1=0x400c2dba, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
Where can I change the size of partition ?
I can't find partitions.csv for esp-idf.

jumjum123, 2020-02-04:
Take a look to
https://github.com/espruino/EspruinoBuildTools/tree/master/esp32/build/app
BTW which version of ESP-IDF is used for your binary ?

bazooka07, 2020-02-05:
Release 3.3 for esp-idf

davidbozo, 2020-02-13:
Same problem here.
Is it some kind of bug or just the binary growed out the partition?
And if i see right in the docs espruino does not support OTA updates. So, could we use the ota partition for the app like this:?
``` csv
#Name,Type,SubType,Offset,Size
#boot,data,0,0x1000,4K
#reserved,0,0,0x2000,24K
#partition,data,0,0x8000,4K
nvs,data,nvs,0x9000,12K
otadata,data,ota,0xC000,8K
free,data,0x40,0xE000,8K
factory,app,factory,0x10000,2688K
flash,data,0x40,0x2B0000,64K
js_code,data,0,0x2C0000,256K
storage,data,0,0x300000,1024K
```
And the other problem is i'm not sure how to achieve this. I found one partition csv under:
`target/esp32/Changes_V3.1/partitions_espruino.csv`
But it looks like it's just a changelog maybe?
As @jumjum123 mentioned there is a **EspruinoBuildTools** which contains another csv but where i can find it? It looks like the `source scripts/provision.sh ESP32` downloads it but here? I'm kind of confused.

jumjum123, 2020-02-14:
I'm sorry, just in vacation on canary island.
Short feedback:
- binary size was growing from version to version. We started somewhere
around 800
- latest version tused was 3.1.3 There have been many changes in later
versions of ESP-IDF. I could not see any changes, which could be helpful
for Espruino. So my decision was follow future versions with low(no)
priority only. May be somebody will give it a try once V4 is released ;-)
- AFAIK, OTA was never implemented/used
- latest csv I used was in https://github.com/espruino/EspruinoBuildTools
- Wilberforce supported buildtools, I did not follow to see what changes he
did. Latest change was switch to ESP-IDF V3.1.3
- provision.sh downloads a lot from buildtools, but no csv. It downloads
compiled .bin for provision

Hope, this helps a bit
Jürgen

davidbozo, 2020-02-21:
No problem thank you for your answer. I was able to build a new partition binary without OTA partition and it works fine.

cascornelissen, 2020-02-21:
@davidbozo, can you describe your steps to successfully flash your ESP32? I'm running into the same problem but I'm not sure how to build a partition binary without an OTA partition.

MaBecker, 2020-02-21:
> No problem thank you for your answer. I was able to build a new partition binary without OTA partition and it works fine.

@davidbozo  would be great if you share your knowledge, because we are short on ESP32 know how, since @wilberforce stoped working for Espruino on ESP32 ( I hope he is fine).

davidbozo, 2020-02-23:
It's maybe a little hacky solution but there is no porblem with it yet.

So, what i could figure out is:
When we run this script: `source scripts/provision.sh ESP32` 
It downloads binarys from this repo: [EspruinoBuildTools](https://github.com/espruino/EspruinoBuildTools/tree/master/esp32/deploy)
This repo contains the **partitions_espruino.bin** and when building espruino it's just copies this file.
So to change the partition table this EspruinoBuildTools project has to be rebuild and overwrite the old **partitions_espruino.bin** in the **app/** folder with the new.
(This 'app' folder will be created after running provision.sh)

To rebuild the [repo](https://github.com/espruino/EspruinoBuildTools):
After cloning it modify this `esp32/build/app/partitions_espruino.csv` file:
``` csv
#Name,Type,SubType,Offset,Size
#boot,data,0,0x1000,4K
#reserved,0,0,0x2000,24K
#partition,data,0,0x8000,4K
nvs,data,nvs,0x9000,12K
otadata,data,ota,0xC000,8K
free,data,0x40,0xE000,8K
factory,app,factory,0x10000,1344K <-- change the 1344K size to 2688K
ota_0,app,ota_0,0x160000,1344K <-- delete this line
flash,data,0x40,0x2B0000,64K
js_code,data,0,0x2C0000,256K
storage,data,0,0x300000,1024K
```
[https://github.com/espruino/EspruinoBuildTools/tree/master/esp32](https://github.com/espruino/EspruinoBuildTools/tree/master/esp32)
Here is some documentation at the bottom of the page. But instead of `source ./build-idf.sh` you can run `source ./build-partition.sh` in the build folder which is enough as far as i remember.

Sorry, this is what i remember and i can't repeat and test the steps only in work.

MaBecker, 2020-03-12:
Ok, let's fix this.

* add firmware size check  like we use in ESP8266

* provide a new `partitions_espruino.bin`

```
#Name,Type,SubType,Offset,Size
#boot,data,0,0x1000,4K
#reserved,0,0,0x2000,24K
#partition,data,0,0x8000,4K
nvs,data,nvs,0x9000,12K
otadata,data,ota,0xC000,8K
free,data,0x40,0xE000,8K
factory,app,factory,0x10000,1536K
free2,data,0x40,0x190000,1536K
#ota_0,app,ota_0,0x190000,1536K
flash,data,0x40,0x310000,64K
js_code,data,0,0x320000,256K
storage,data,0,0x360000,896K
```

cascornelissen, 2020-03-12:
How fortunate... Decided I wanted to spent some time on this this evening and I just noticed the comment and commits to [`EspruinoBuildTools`](https://github.com/espruino/EspruinoBuildTools) by @MaBecker. Pulled and build everything once more with the new changes and Espruino on my ESP32 is now working.

Thank's a lot for the changes @MaBecker, you rock! ❤️

MaBecker, 2020-03-12:
Cool, so you can build and it works for you, thanks for testing!

there are just two more things missing `boards/ESP32.py` and  `make/targets/ESP32.make` just working on it.

cascornelissen, 2020-03-12:
Yeah, it's building for me without any additional changes (I actually removed everything and cloned the repository again just to check) and the error in the original post that I was running into before is gone on my side. Ran a small script to test if that works as well and that seems to be good too 👍

MaBecker, 2020-03-12:
Just created a pr for the additional required updates

https://github.com/espruino/Espruino/pull/1776/commits/433708fd8fb7f577e493f209e72306e2632c5cd9

MaBecker, 2020-03-12:
I guess it will not pull down the new app.tgz version

https://github.com/espruino/Espruino/blob/433708fd8fb7f577e493f209e72306e2632c5cd9/scripts/provision.sh#L76-L80

@gfwilliams any hints how to handle updates of app.tgz ?

gfwilliams, 2020-03-13:
Is the app.tgz for a new SDK? If so you could create as new branch (like was done for 3.1)?

Or just create a new file - `app.3.1.tgz` vs `app.tgz`

MaBecker, 2020-03-13:
Still the same version , but with a new binary partition file.

gfwilliams, 2020-03-13:
Ok, I'd just update it. I know the whole build/version thing is tricky but presumably all it means it it *was* broken and now it's working?

jumjum123, 2020-03-13:
It was broken during latest changes (fonts need more memory?)
Size of binary become more than expected for partition table.
So we had to resize partition size for application
Short version, fixed partition size for application to have space for
latest extensions

MaBecker, 2020-03-13:
The issue we are talking about is not the build process. The firmware did not fit into the partition.

To be 100% save, I have to flash a Travis build tonight.

MaBecker, 2020-03-13:
just test, works like never was broken :)

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v04.401 (c) 2019 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>process.env
={
  VERSION: "2v04.401",
  GIT_COMMIT: "e568c6c7",
  BOARD: "ESP32",
  FLASH: 0, STORAGE: 262144, RAM: 524288,
  SERIAL: "a4cf1276-64b4",
  CONSOLE: "Serial1",
  MODULES: "Flash,Storage,hea" ... "r,crypto,neopixel",
  EXPTR: 1073484860 }

```

closed 2020-03-13 by MaBecker

HyGy, 2020-03-16:
wifi.save() does not work:

```
=function () { [native code] }
>wifi.connect('hygywlanking', {password: 'nEgrO111'}, function() {
:    console.log('Connected to Wifi.  IP address is:', wifi.getIP().ip);
:    wifi.save(); // Next reboot will auto-connect
:});
ERROR: Wifi: event_handler STA_START: esp_wifi_connect: 12298(SSID is invalid)
=undefined
WARNING: Wifi:startMDNS - espressif
Connected to Wifi.  IP address is: 192.168.1.219
Uncaught Error: File already written with different data
 at line 2 col 15
    wifi.save();
              ^
in function called from system
>wifi.save();
Uncaught Error: File already written with different data
 at line 1 col 11
wifi.save();
          ^
>
```

GaryOtt, 2022-10-21:
I have been experiencing severe networking problems with the ESP32 builds in recent months and quite simply have to overcome them.

I just wanted to share that I am working on migrating the ESP32 version to ESP-IDF 3.3.6 which was released end of January this year. It has a lot of WiFi related updates in the change log. I have compiled and flashed several of the example projects and not found any problems so that's hopeful but not definitive. I hope this maybe an easier job than migrating to ESP-IDF 4.x and more likely to succeed. I may look at a 4.x upgrade later (definitely so if 3.3.6 doesn't fix the problems).

The specific symptom I was experiencing was a failure to pipe data to a TCP socket because the socket gets stuck in a drain wait condition for tens of seconds, sometimes minutes, sometimes seemingly infinitely. My investigations confirmed the root of the problem is within ESP-IDF.

Something that surprised me is that the client machine has an impact upon the problem. My macOS development machine that rarely had a problem can no longer open the web interface served by my Espruino based project following an OS update. I was first aware of problems on some Windows devices and then iOS devices. A machine running Linux Mint has had no problems.

So wish me luck and keep an eye out for updates!