espruino/Espruino #1758: Bangle.js: New devices have wrong LCD gamma

gfwilliams, 2020-02-14:
I'm pretty sure the Node.js/Bangle Beta devices have a slightly different LCD to the one I use for testing - the contrast is *way* off, and we need new gamma curves.

Here's my testing:

```
for (var i=0;i<240;i+=2) {
  var c = i/240;
  g.setColor(c,c,c);
  g.fillRect(i,0,i+1,239);
}
var y;
// random dots to give some idea of what gamma SHOULD be
for (var i=0;i<240;i++) {
  var c = i/240;
  for (y=100;y<140;y++) {
    g.setPixel(i,y,(c<Math.random())?0:-1);
  }
}

function setGamma(gamma) {
  Bangle.lcdWr(0xe0,gamma); // PVGAMCTRL
  Bangle.lcdWr(0xe1,gamma); // NVGAMCTRL
}

Bangle.setLCDPower(1);
Bangle.setLCDTimeout(0);
//BETTER setGamma([0xf0,0x09,0x13,0x0A,0x0b,0x06,0x38,0x33,0x4f,0x04,0xd,0x19,0x2e,0x2f]);
//BAD - setGamma([0xd0,0x06,0x0c,0x09,0x09,0x25,0x2e,0x33,0x45,0x36,0x12,0x12,0x2e,0x34]); // what we use now
//BETTER setGamma([0x70,0x2C,0x2E,0x15,0x10,0x09,0x48,0x33,0x53,0x0B,0x19,0x18,0x20,0x25]); // https://github.com/Xilinx/linux-xlnx/blob/master/drivers/staging/fbtft/fb_st7789v.c

// BAD - https://forum.arduino.cc/index.php?topic=546228.0
//Bangle.lcdWr(0xe0,[0xD0, 0x00, 0x05, 0x0E, 0x15, 0x0D, 0x37, 0x43, 0x47, 0x09, 0x15, 0x12, 0x16, 0x19]);
//Bangle.lcdWr(0xe1,[0xD0, 0x00, 0x05, 0x0D, 0x0C, 0x06, 0x2D, 0x44, 0x40, 0x0E, 0x1C, 0x18, 0x16, 0x19]);

// best so far - altered curve
// index 1 onwards changes the lighter end of gamma
// last array elements change the darker end of gamma
setGamma([0x70,0x1f,0x20,0x15,0x10,0x09,0x48,0x33,0x53,0x0B,0x19,0x15,0x2a,0x2f]);
```

So the last command should really improve contrast

gfwilliams, 2020-02-14:
Actually the data is described in `8.19 Gamma Correction`  and `PVGAMCTRL` in https://www.newhavendisplay.com/appnotes/datasheets/LCDs/ST7789V.pdf

But randomly tweaking numbers may be easier!

MaBecker, 2020-02-14:
I can confirm this, the first watch is much darker than the two I got via Kickstarter.

MaBecker, 2020-02-14:
No sure, does this difference has to do with gamma, I would say no ...

The left one is from kickstarter , it is much brigther, both have set `Bangle.setLCDBrightness(1)`

[image]

gfwilliams, 2020-02-24:
I have a watch with a very dim LCD as well that I ordered from eBay. I believe if this was a pre-kickstarter watch, they were likely using up old stock and may have made some with LCDs that had dodgy backlights

MaBecker, 2020-02-24:
>  I believe if this was a pre-kickstarter watch, they were likely using up old stock and may have made some with LCDs that had dodgy backlights

yes it’s from the same UK  eBay reseller :)

I am glad that the kickstarter one’s are much brighter.

closed 2020-03-03 by gfwilliams

MaBecker, 2020-03-05:
Wow, this the death for the right one..... wrong colours and very dark.

would like to add some ifndef else endif

What do you think?

gfwilliams, 2020-03-05:
I doubt many people would roll their own firmware... It'd be trivial to use the example code here to make an app that wrote the new gamma values at boot though?

MaBecker, 2020-03-05:
Thank, good point, so that's what I do.

gfwilliams, 2020-03-06:
When you get the right code for it, ensure the app writes it into `.boot1` (the bootloader writes it to `.boot0` if you want an example). That way it'll run regardless of whether the app uses widgets or not

commit ca6e4731 (2020-03-03): Bangle.js: Tweak LCD gamma curves to provide better gradients on new watches (fix #1758)