espruino/Espruino #1760: nrf52840dk support

glococo, 2020-02-18:
Hi everyone,

I found your project awesome, and I would like to use my nordic dev board with Espruino.
I tried to compile using "make clean && BOARD=NRF52840DK RELEASE=1 make" but failed 
`error: 'UARTE1_EASYDMA_MAXCNT_SIZE' undeclared (first use in this function); did you mean 'UARTE0_EASYDMA_MAXCNT_SIZE'?`

Is there any bootloader / hex available for this board ? or how can I fix this compile issue ?
Thanks in advance.

glococo, 2020-02-18:
Hi, I modified 
`#define UARTE1_LENGTH_VALIDATE(...)  UARTEX_LENGTH_VALIDATE(UARTE1, __VA_ARGS__)`
to
`#define UARTE1_LENGTH_VALIDATE(...)  UARTEX_LENGTH_VALIDATE(UARTE0, __VA_ARGS__)`

and compiled, but I actually don't know what I did :(

btw: The hex worked fine in my pca10056 nrf52840-DK board. Now playing :)

gfwilliams, 2020-07-14:
Great - glad it's working. Some changes went in the last few months to make the nRF52840 build work much better as well

closed 2020-07-14 by gfwilliams