espruino/Espruino #1765: Unexpected reset

DVR-Ltd, 2020-03-02:
While trying to diagnose the cause of an unrelated issue (probably in my code), I wrote code to consume memory until only 2000 JSVars remained free. I found that my debug code revealed a bug in Espruino. Below is a minimal example of the problem. The code will cause an Espruino WiFi to freeze and an ESP32 to reset. Both boards are running Espruino v2.04.

```
var linkList  = {
		val:  12345.12345,
		next: null
	},

	last      = linkList,

	consumeMemory = () => {
		var memory = process.memory();

		last.next = {
			val: 12345.12345,
			next: null
		};

		last = last.next;

		console.log("memory.free: ", memory.free);

		if (memory.free > 2000) {
			setTimeout(consumeMemory, 0);
		}
		else {
			console.log("Bug not replicated.");
		}
	};

consumeMemory();
```

gfwilliams, 2020-03-02:
Thanks! This is really interesting.

I'll look into it but first thoughts are that the Garbage collector is (partially) recursive. It's possible that by creating a massive linked list you're effectively causing the garbage collector to recurse and it's not checking the stack level to ensure it doesn't overflow.

DVR-Ltd, 2020-03-02:
That makes a lot of sense. This code probably makes the worst case scenario for the garbage collector. Assuming you're right, a fix may incur a performance hit that can't be justified by eliminating an unexpected behaviour in an edge case.

I rewrote the code to be less obnoxious. It now pushes an object onto the end of an array instead of creating a linked list which worked fine.

commit 8bb747f2 (2020-10-08): Check for Stack overflow when Garbage Collecting giant linked list (fix #1765)
closed 2020-10-08 by gfwilliams