espruino/Espruino #1766: Add Storage check at startup - erase all if storage looks corrupt

gfwilliams, 2020-03-03:
This came about after a recent firmware update that increased the file name length for `Storage`: http://forum.espruino.com/conversations/344251/#comment15131232

gfwilliams, 2020-03-06:
A few people having issues updating - so I think this one needs doing

conoro, 2020-03-06:
Would this also be responsible for "Delete All Apps" failing on every watch I've upgraded in the past week? So I have the watch running old apps and old (month+) firmware:

1. Update Firmware
2 Connect from App Loader 
3. Go to About > Delete All Apps
4. Shows activity bar for 20 or 30 seconds
5. Always gives an error. 
6. Click "Install Default Apps" 
7. The deletion phase of that works fine and the new apps then all install ok. 

So it's not a blocker, it's just confusing users that the first delete never works.

gfwilliams, 2020-03-06:
If you're using the old firmware, does https://banglejs.com/oldapps/ work for deleting all the apps?

https://banglejs.com/apps seems to work fine for deleting apps for me on a new Bangle.js firmware.

This issue won't be a problem for running old firmware. It's only if you update an old firmware to new firmware and the watch then doesn't load apps and becomes hard to connect to

conoro, 2020-03-06:
Ah gotcha thanks.

This worked ok:

1. Update Firmware
2. Remove all apps, using /oldapps
3. Install default apps using /apps

commit a80403c9 (2020-03-31): Add Storage check at startup and erase everything if storage looks corrupt (fix #1766)
closed 2020-03-31 by gfwilliams