espruino/Espruino #1767: Bangle.js and FW 378 doesn't boot

ra100, 2020-03-04:
I've flashed espruino_2v04.378_banglejs.zip to bangle.js but after restart it stays on Bangle.js logo, and only thing I can do is to flash some other FW.
Flashing to .85 works fine.

gfwilliams, 2020-03-04:
Have you tried installing completely new JS from https://banglejs.com/apps/ (about -> install default apps).

In recent versions I made some big changes to Storage, so previously uploaded apps would be lost.

ra100, 2020-03-04:
So I should first install new apps and then update the firmware?

gfwilliams, 2020-03-04:
No, you should be able to just install the firmware and the reflash.

Did you try this: https://www.espruino.com/Bangle.js#resetting-without-loading-any-code

It'll stop Espruino from loading any files at boot, which should allow you to program it just fine.

Worst case you could do `require("Storage").eraseAll()` on the old firmware, and then write new Bangle.js firmware - but that shouldn't be required

ra100, 2020-03-04:
Ah, ok, now I get it, thanks.

closed 2020-03-04 by ra100

gfwilliams, 2020-03-04:
I think this would probably be fixed by https://github.com/espruino/Espruino/issues/1766 when implemented too