espruino/Espruino #1770: improvements for the  new bitmap_font_6x8.c

MaBecker, 2020-03-06:
Some chars might look better when the are centered, eg  ! " , ; : ` ( )  |

part 1  has correct values

[image]

part 2 needs some corrections, will come up with a list

[image]

```
// ugly code but does it's work
function drawCharTable(p) {
    var cc = 33, sf = 2, x = 0, y = 50, yo = 1, i = 33;
    s = p ?  8 : 2;e = p ? 16 : 8; yo = 1;
    g.clear().setFont("6x8",sf);
    g.drawString("part "+(p+1), 80, 20);
    g.drawString(" 0123456789ABCDEF",x,y);
    for (i = s;i<e;i++){ g.drawString(i.toString(16).toUpperCase(),x,y+8*sf*yo);yo++;}
    cs = p ?  128 : 32; ce = p ? 255 : 127; yo = 1;x =0;
    for (row = 0; row < 16; row++) {
        sfcc = " ";
        console.log(cs,ce);
        if (cs > ce) break;
        for ( col = 0; col < 16;col++ ) {
            sfcc += String.fromCharCode(cs);
            cs++;
        }
        g.drawString(sfcc,x,y+(row+1)*8*sf);
    }
}

//drawCharTable(0);
drawCharTable(1);
```

gfwilliams, 2020-03-06:
Ok, thanks! The font itself is ISO8859-1/windows-1252 (which I believe is actually the 'default' 8 bit encoding for web browsers) whereas the Dylex font used ISO10646 (which I believe is UTF-16 - so just the bottom-most 256 characters of the 0-65535 range).

Since `String.charCodeAt` returns UTF-16, simple translation from the web browser would be using the ISO10646 codepage.

So it looks like we could just convert the ISO8859-1 font to ISO10646-1, and there may be ways to programatically do this (which might save you some hard work!)

gfwilliams, 2020-03-06:
Actually: https://www.cl.cam.ac.uk/~mgk25/unicode.html#ucs

> What are UCS and ISO 10646?
> The UCS characters U+0000 to U+007F are identical to those in US-ASCII (ISO 646 IRV) and the range U+0000 to U+00FF is identical to ISO 8859-1 (Latin-1)

So it may be that the character positions are actually ok!

MaBecker, 2020-03-06:
> So it may be that the character positions are actually ok!

I think so too , but it’s a big difference in look and feel comparing old with new font especially for clocks

gfwilliams, 2020-03-06:
> it’s a big difference in look and feel comparing old with new font especially for clocks

Yes... However the character spacing should at least be the same.

Talking of which I guess it might be nice to add a `6x8v` mode where certain 'thin' characters were actually rendered thin.

MaBecker, 2020-03-06:
[image]

MaBecker, 2020-03-06:
Do you see the spacing

gfwilliams, 2020-03-06:
Of the colon? Yes, that needs to change. Stuff like that should be pretty easy to tweak with: https://github.com/espruino/Espruino/blob/master/libs/graphics/bitmap_font_6x8.c

MaBecker, 2020-03-06:
> Of the colon? Yes, that needs to change

Yes, this is what I mean with:

> Some chars might look better when they are centered, eg ! " , ; : ` ( ) |

gfwilliams, 2020-03-06:
See most recent update ^^^

MaBecker, 2020-03-06:
Wow, you are to fast for me 😁

MaBecker, 2020-03-08:
> So it may be that the character positions are actually ok!

* Compared codes and chars with https://ascii.cl/htmlcodes.htm just a few changes are needed

* Circumflex or accent circonflexe (â, ê, î, ô, û) are very hard with 6x8 https://en.wikipedia.org/wiki/French_orthography

*  add char code values 

* update ` : "\_XX__"

* update ´ :  "__XX\_"

* update ^ : "_XXX_"

Have you thought about  a translation for values higher than 255, like 338  to 140?

MaBecker, 2020-03-09:
latest version, might help to check

[image]

closed 2020-03-09 by MaBecker

gfwilliams, 2020-03-09:
> Have you thought about a translation for values higher than 255, like 338 to 140?

The issue is that Espruino Strings can't support anything outside the 0-255 range. If there are unused characters then sure, we can use them for other stuff - but it'd need to be documented because if nobody knows they are there it's not useful :)

Perhaps this now means that the unicode converter in the IDE should stop doing the strange UTF-8 stuff, and should just convert to ISO 10646 (first 8 bits). I could update the terminal to work in a similar way as well.

MaBecker, 2020-03-09:
I had a lookup table in mind to handle this.

gfwilliams, 2020-03-09:
Where would the lookup table be though? In the Web IDE before upload? I'd have to add it to BangleApps as well...

MaBecker, 2020-03-09:
Not sure, but maybe  this is too much effort for less than a hand full of chars ....

And just keep it in mind to add it to the Fonts documentation page.