espruino/Espruino #1773: Bangle.js E.showMenu(menu); glitch an improvement

sebi5361, 2020-03-07:
I find `E.showMenu(menu);` great.

I have noticed a small glitch: if the up and bottom arrows are displayed on the right, if a text line is too long, its last character will mess up the last column of the screen when scrolling up and down.

Additionally a nice improvement would be to scroll long lines automatically to the left when highlighted, so that they can be completely read, don't you think?

This feature could also exist in the launcher where it is difficult to read app names as they are truncated on the right.

gfwilliams, 2020-03-09:
Sounds like a plan as long as it doesn't slow down the menu rendering too much for normal every-day cases.

I'm open to PRs ;) https://github.com/espruino/Espruino/blob/master/libs/js/banglejs/E_showMenu.js

sebi5361, 2020-03-10:
Indeed it might slow down the rendering... so not worth it.

And to avoid such a glitch:
[image]
one just has to take care that the text doesn't exceed 19 characters or use `.substring(0, 19);`.
Am closing this issue.

closed 2020-03-10 by sebi5361