espruino/Espruino #1775: Include Android.h to build

B1tb0t, 2020-03-10:
I am currently including my own C module in the espruino firmware. This code is adapted from an arduino library and it needs the arduino.h file. I followed this tutorial: https://github.com/espruino/Espruino/tre­e/master/targets/esp32 , but the build failes with: "fatal error: Arduino.h: No such file or directory". Am I doing it right?

Jonas

jumjum123, 2020-03-10:
This was helpful in a first step for Espruino, and was used to get http
running that time.
Later it was removed totally and never used again.
May be you can find something in https://github.com/espressif/arduino-esp32

Regards from Krefeld(cloudy, rainy about 10 degrees Celsius)
Jürgen

B1tb0t, 2020-03-11:
I successfully included the arduino-esp32 addon in a new ESP-IDF project (template). But how do I include it in the espruino build ? Copying it into the "esp-idf/components" folder in the espruino project dont work. There is also no make file to build the components.

Best regards from Wuppertal 
Jonas

jumjum123, 2020-03-11:
Sorry, but I've no idea, how you build Espruino.
My (short) path is:
- create development for esp-idf described in documentation from Espruino
- clone esp-idf-template in its own folder, and replace partition and
sdkconfig
- if you implement your own arduino-whatever, add this to components
- run make for esp-idf-template
- create tgz files for app and esp-idf, see espruino buldtools in guthub
- clone Espruino from github
- add content of tgz files above to Espruino source
- add your sources to Espruino folder
- add necessary lines for include etc. to Espruino/make/family/esp32.make
- run make for Espruino

Obviously, this does not explain each step in detail.
Anyway, it should be help to help yourself

Regards
Jürgen
-

gfwilliams, 2020-07-14:
Closing this - IMO including Arduino libs *inside* Espruino is a super big undertaking. Realistically even if you can get it to compile, it'll probably break a lot of Espruino by reconfiguring peripherals behind the scenes when Espruino expects them to be something else.

Of course if you're including `Arduino.h` just for digitalWrite/etc then it's best to write your own file:

```
void digitalWrite(int pin, int value) { jshPinOutput(pin,value); }
// and so on...
```

closed 2020-07-14 by gfwilliams