espruino/Espruino #1779: How to compile a firmware for ESP32 that can handle SD Cards

MaBecker, 2020-03-16:
```
/* jshint esversion: 6 */
//File_IO_and_SD_cards.js 

if ( process.env.BOARD == "ESP32") {

    M5Stack = require("M5STACKS");
    SPI1.setup({mosi:M5Stack.MOSI, miso:M5Stack.MISO, sck:M5Stack.SCK});
    E.connectSDCard(SPI1, M5Stack.CS);

    // see what's on the device
    //console.log(require("fs").readdirSync());
}

if ( process.env.BOARD == "PICO_R1_3") {

  // Wire up up MOSI, MISO, SCK and CS pins (along with 3.3v and GND)
  SPI1.setup({mosi:B5, miso:B4, sck:B3});
  E.connectSDCard(SPI1, B6 /*CS*/);
 
  // see what's on the device
  //console.log(require("fs").readdirSync());

}
```

No error on Pico, but this one on ESP32.

```
Uncaught Error: Unimplemented on Linux
 at line 5 col 37
    E.connectSDCard(SPI1, M5Stack.CS);
```

Are there some modules or -DSettings for ESP32 missing?

https://github.com/espruino/Espruino/blob/e568c6c7ac78e81b4ab40bb28bd92e8052f4e7b6/libs/filesystem/jswrap_file.c#L145-L162

MaBecker, 2020-03-17:
Wow the solution was there in the same file just a few lines above :)

https://github.com/espruino/Espruino/blob/e568c6c7ac78e81b4ab40bb28bd92e8052f4e7b6/libs/filesystem/jswrap_file.c#L22-L24

So the solution is to remove  this line 

https://github.com/espruino/Espruino/blob/e568c6c7ac78e81b4ab40bb28bd92e8052f4e7b6/boards/ESP32.py#L35

and build a new firmware version and it works.

closed 2020-03-17 by MaBecker

AkosLukacs, 2020-03-18:
Just a hunch: does writing to internal flash still work?

gfwilliams, 2020-03-18:
Using Storage it would - FLASHFS is something different (FAT32 on the internal flash)

AkosLukacs, 2020-03-18:
That's great news, thanks for the clarification.
Btw, is there a list of boards.py or common compiler flags?