espruino/Espruino #1786: Pretokenise doesn't handle RegExp literals correctly

undecided, 2020-03-26:
Hi! Got my banglejs at the beginning of the week, been loving it!
Have noticed a small issue when trying to write programs on https://www.espruino.com/ide/ with regular expressions:
```
var foo = function(regex) { 
  if(regex) { 
    return ['foo.js', 'foo.info'].filter(function(x) { if(x.match(regex)) { return x; } });
  } 
};
var fileList = function(regex) { return foo(regex); };
var appList = function() { return fileList(/\.info$/); };

console.log(fileList(/\.info$/));

console.log(appList());
```
(edited to something that has data to filter on the emulator version)

So I would expect both these `console.log`s to return the same array - looks like the following code is being generated:

```
function fileList(regex) {return storage.list(regex)}
function appList(undefined) {return fileList(REGEX)}
```
(as produced by `dump()`) but I can't see REGEX being defined anywhere. Don't have minification enabled, so I guess it just the transpiler at fault. 

Thought it would be easy to workaround with old-style anonymous functions - but they do exactly the same! Any thoughts?

(Note: This does not seem to have the same problem on https://www.espruino.com/ide/emulator.html - I have no idea why...)

gfwilliams, 2020-03-26:
Just wanted to add - thanks for being so clear with the explanation. It helps a massive amount with narrowing down the actual issue.

opichals, 2020-03-26:
> It might be worth turning off pretokenise for Bangle.js now - now we're executing from flash there's not much benefit.

Wonder if there is performance benefit in there. 1 char vs a `function` sounds like there could.

gfwilliams, 2020-03-26:
There is - but it doesn't pretokenise when executing from flash (because then that'd use RAM for the tokenised function).

I've been considering enabling pretokenisation in the Bangle app loader though - that would definitely improve performance because the upload would be faster *and* code would execute faster

undecided, 2020-03-26:
Ahhh - that's solved the problem!

```
>E.setFlags({pretokenise:0})
=undefined
>function foobar() { console.log(/AAA/) }
=function () { ... }
>foobar()
RegExp: {
  "source": "AAA",
  "lastIndex": 0 }
=undefined
>E.setFlags({pretokenise:1})
=undefined
>function foobar() { console.log(/AAA/) }
=function () { ... }
>foobar()
RegExp: {
  "source": "",
  "lastIndex": 0 }
=undefined
```

> Just wanted to add - thanks for being so clear with the explanation. It helps a massive amount with narrowing down the actual issue.

No problem, many thanks for being so responsive! Espruino is a phenomenal platform, I've been a very happy customer since I got my first few picos! Enormously impressed by everything you've managed to achieve here </gush>

commit 76330371 (2020-03-28): Fix pretokenise of RegExp literals (fix #1786)
closed 2020-03-28 by gfwilliams