espruino/Espruino #1787: ESP32 espruino_2v04 | Replay "What?" or "How?" on any command

calkoe, 2020-03-26:
Hi,
my ESP32 always reply "What?" or "How?" to JS commands.
Espruino 2.04 on my ESP8266 works just fine.

any suggestions ?

```
calvin@Calvins-MBP Espruino % espruino -b 112500
Espruino Command-line Tool 0.1.31
-----------------------------------

Using first port, {"path":"/dev/tty.SLAB_USBtoUART","usb":[10,null],"description":"Silicon Labs"}
Connecting to '/dev/tty.SLAB_USBtoUART'
No result found for "process.env" - just got "\u0010print(\"<\",\"<<\",JSON.stringify(process.env),\">>\",\">\")\r\nHow?\r\n>"
Unable to retrieve board information.
Connection Error?
Connected
console.log("Hello World!");
What? 
>console.log("Hello World!");
What? 
>let a = 3;
What? 
>a
What? 
>
```

closed 2020-03-26 by calkoe

gfwilliams, 2020-03-26:
Any explanation?

calkoe, 2020-03-26:
I figured it out myself, 
i have not uploaded all files as described here:
http://www.espruino.com/ESP32#firmware-updates

Now it works!  :)