espruino/Espruino #1791: espruino crashes due to an invalid read

strongcourage, 2020-04-01:
Hi,

I found an invalid read of espruino (the latest commit a80403c) when it calls the third-party library TensorFlow. This bug causes program to crash. Actually the crashing input is generated by modifying 2 bytes of tests/test_tensorflow.js.
~~~
173c173
<     0x�5, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00,
---
>     0x75, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00,
191c191
<     0x65, 0x6e, 0x74, 0x69, 0x61, 0x6c, 0x5f, 0x31, 0x2f, 0x64, 0x65, 0x6e<
---
>     0x65, 0x6e, 0x74, 0x69, 0x61, 0x6c, 0x5f, 0x31, 0x2f, 0x64, 0x65, 0x6e,
~~~

PoC: https://github.com/strongcourage/PoCs/blob/master/Espruino_a80403c/ir_poc

Valgrind says:
~~~
==16054== Invalid read of size 2
==16054==    at 0x733222: GetOptionalFieldOffset (flatbuffers.h:2264)
==16054==    by 0x733222: GetField<unsigned char> (flatbuffers.h:2268)
==16054==    by 0x733222: is_variable (schema_generated.h:2890)
==16054==    by 0x733222: tflite::MicroAllocator::AllocateTensors() (micro_allocator.cc:106)
==16054==    by 0x727467: tflite::MicroInterpreter::AllocateTensors() (micro_interpreter.cc:147)
==16054==    by 0x84CFC3: tf_create (tensorflow.cc:69)
==16054==    by 0x6F5095: jswrap_tensorflow_create (jswrap_tensorflow.c:72)
==16054==    by 0x4AC151: jsnCallFunction (jsnative.c:217)
==16054==    by 0x4BA34C: jspeFunctionCall.constprop.11 (jsparse.c:597)
==16054==    by 0x4BB47B: jspeFactorFunctionCall (jsparse.c:1165)
==16054==    by 0x4BBB28: jspePostfixExpression (jsparse.c:1769)
==16054==    by 0x4BFD5A: jspeBinaryExpression (jsparse.c:1932)
==16054==    by 0x4BFD5A: jspeConditionalExpression (jsparse.c:1968)
==16054==    by 0x4BFD5A: jspeAssignmentExpression (jsparse.c:2027)
==16054==    by 0x4BFD5A: jspeStatementVar (jsparse.c:2142)
==16054==    by 0x4CA2DC: jspeBlockOrStatement (jsparse.c:2101)
==16054==    by 0x4CA4EA: jspParse (jsparse.c:2113)
==16054==    by 0x4CD6FE: jspEvaluateVar (jsparse.c:2964)
==16054==  Address 0xffffffffddd57388 is not stack'd, malloc'd or (recently) free'd
~~~

Best,
MD

gfwilliams, 2020-05-11:
Thanks - this seems like an issue that should be reported in Tensorflow itself, since it is that library which fails? I know their file parsing seems particularly unsafe (but probably because they're not expecting anyone to be able to push arbitrary models to it).

strongcourage, 2020-05-11:
I agree with you, so just for your information. Thanks.

closed 2020-05-11 by strongcourage