espruino/Espruino #1796: Graphics.fillPoly glitch

nebbishhacker, 2020-04-12:
`g.fillPoly` sometimes produces extraneous horizontal lines when two adjacent points in the list are almost, but not quite, horizontally aligned.

Test case: 
```
g.clear();
g.fillPoly([120, 40,
            200, 200,
            170, 200,
            120, 100,
            70, 180, // line starts at this point and extends to the right
            40, 179]);
```

Result:

[image]

Bug was discovered while working on an SVG to bangle draw call converter :)

gfwilliams, 2020-04-14:
Thanks - we recently 'improved' the fillPoly algorithm so it looks like there's still something odd here.

Just for reference, here's the outline:

![](data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAPAAAADwCAYAAAA+VemSAAAMqUlEQVR4Xu2c23KUOQwGhyou8v5PywVVuxWWsGGSyfgguyX/zS22Prml5gf28O12u/1z84cEJFCSwDcFLjk3m5bALwIK7CJIoDABBS48PFuXgAK7AxIoTECBCw/P1iWgwO6ABAoTUODCw7N1CSiwOyCBwgQUuPDwbF0CCuwOSKAwAQUuPDxbl4ACuwMSKExAgQsPz9YloMDugAQKE1DgwsOzdQkosDsggcIEFLjw8GxdAgrsDkigMAEFLjw8W5eAArsDEihMQIELD8/WJaDA7oAEChNQ4MLDs3UJKLA7IIHCBBS48PBsXQIK7A5IoDABBS48PFuXgAK7AxIoTECBCw/P1iWgwO6ABAoTUODCw7N1CSiwOyCBwgQUuPDwbF0CCuwOSKAwAQUuPDxbl4ACuwMSKExAgQsPz9YloMDugAQKE1DgwsOzdQko8OE78OPnj9vL95fDX3nd5ynwwbN/k1eJzx2yAp8725sCHzzc309T4ENnfP/V9St85qAV+My5/vn6vj1Pgc8ctAIfONdHsirxecNW4PNm+uHr61f4wCH7Z+Azh/rsK/vs58+kcu6r/AIfNNsWOVvOHITk+Kco8EEjbpWz9dxBaI59igIfMtoeKXvOHoLn2Gco8CGj7ZWy9/whmI57hgIfMNIRGUfuHIDquCco8AEjHZVx9N4ByI55ggIXH+WMhDN3i2M7pn0FLjzKCAEjahRGWL51BS48wgj5ImoURli+dQUuOsJI8SJrFcVZtm0FLjq6SOkiaxXFWbZtBS44uhXCrahZEG25lhW43MhuD/9ro5mnKPAMPe6uAnPsh5JXiray9tBjvfSUgAI/RZTnwGrBVtfPQ/KcThS40Cx3CLYjoxDy9K0qcPoR/dfgLrF25RTBnr5NBU4/or0C7/zFogj61G0qcOrx7JdXgQssxLsWFTj5vKjf0lK5yceRrj0FTjeSvxuiRKJyk48jXXsKnG4k/zdES0TnJx5NmtYUOM0oPjZCC0TnJx5NmtYUOM0ocvzW+R6HEiddkN9tKXDS+WQRJ0sfSceEt6XA+Ajy/dbZr3DCpXjQkgInm1XGL17GnpKNDWtHgTH0nwdnlSVrX8nGt70dBd6O/HFgZkky95ZohNtbUeDtyGsK/Nq1EidaFv8WOtcwKshRocdcU13fjV/g9YybEqrIUaXPJugHHFLgBEOsJEWlXhOMdnkLCrwc8dcBFYWo2DM85mXxCrwMbVvhijJU7LltGvVOKTA4s8oiVO4dHHl4tAKHI20vWFmCyr23Tyj/SQWGZnSCACe8ARp/WKwCh6HsK3TC8p/whr6p5TutwMBMTlr8k94CrMJ0pAJPI+wrcNrCn/aevmnypxV48wxOXPgT37R5LYbjFHgYXf/FUxf91Hf1T3j/DQXeyPzkRT/5bRtXpDtKgbuRjV04fcFPf9/Y1NffUuD1jH8lXGHBr/DGTevSHKPAzajGD15lsa/yzvFNiL+pwPFM/6p4taW+2nsXr8/T8gr8FNHcgast9NXeO7cd87cVeJ7hwwpXXearvnvhKj0srcALqV91ka/67oWrpMC74V59ia/+/l375hd4EeksC0z1QeUuGmfasgq8YDRZlvetD6ofKnfBSNOWVODg0WRaWgUOHm7CcgocPJQsAt/3QfVF5QaPNW05BQ4cTaZlVeDAwSYupcCBw8ki8KM+qP6o3MDRpi2lwEGjybKkX/VB9UjlBo02dRkFDhpPliV91seznw/C8aEMlbvqPVnqKnDAJLIsZ0sfLWcCkCjwCoif1FTgSdCUEPdt9/TRc3YSz1/XqdzIN2SrpcCTE8mylD199JydxKPAkQD9AsfSpESY+fq+3aV6p3JjJ5+nml/giVlkWcaRPkbuTKD6c5XKjeg9Yw0FHpxKlkWc6WPm7iC2X9eo3Jmes95V4MHJZFjC2R5m7w+iU+BRcP4ZOIYctfgRf/ZdUWOEahaGI71nuuMXuHMaWRYvqo+oOp0Y/Qr3AntwXoE7QVIL/77N6B6i67UipXJb+6twToE7ppRl4aL7iK7XipTKbe2vwjkF7phShoVb1cOqus/wUrnP+qry8wrcOKksi7aqj1V1n+Glcp/1VeXnFbhhUlmWbHUfq+s/Qk3lNow+/REFbhhRhgXb0cOOjM9wU7kNo09/RIGfjCjLcu3qY1fOPXYqN72hTxpU4AIC71zunVkr/9FYdTFb+1fgL0hRy0wvNvVuKrdVloznFFiBPxCgRKJyM4rZ2pMCPyCVYZnIHqhsKrdVmGznFPiTiWRZIrIPKpvKzSZmaz8KnFTgDItM9UDltkqT6ZwC300jw/Jk6OEVC9UHlZtJzNZeFFiBv9wVSiYqt1WcLOcU+N0kMixNhh7of4xFfv2ziNnahwInEjibvG9oqL6o3FZ5MpxT4N9TyLAsGXr4bCmpvqjcDGK29qDA4F/WZPitauuiUDJRua1c6HMKrMBNO0iJROU2QUlw6PICZ1iQDD207CLVJ5XbwoQ+o8A/f9xevr9gc6i0nFSvVC62FB3BlxY4w2Jk6KFjX/yXO3pgbTirwH59u9aM+gWHyu2CAxy+rMD0QtD5M7tG9U7lzrBaffeSAmdYhAw9jC4X1TuVO8ppxz0F3kH5LuOERaTeQOUCa9IUeTmBMyxAhh6atuOLQ9QbqNxZXqvuK/Aqsg/qnrSA1Fuo3M2r0hR3KYHpwdP5TRvRcYh6D5XbgWbbUQXehpr7D+RXPpGSicpdyXKk9mUEpgdO548sR8sd6l1UbguTnWcuITA9bDp/9UJR76NyV/Psqa/APbQGz56+aNT7qNzBNVhy7XiB6SHT+Uu25pOi1Dup3F1cn+Uo8DNCkz9/lQWj3knlTq5F2PWjBaaHS+eHbUljIeq9VG4jlqXHFHgR3isuFfVmKnfR6nSVPVZgeqh0ftcWBB6m3k3lBqIbKnWkwPQw6fyhTQi6RL2dyg3CNlxGgYfRfX7xqov0ngbFgMoNXqGucscJTA+Rzu+a/qLDFAMqdxHGprIK3ISp7dAVF+gRGYoFldu2IfGnjhKYHh6dH78e4xUpFlTuOKm5mwo8x+/P7astTgs2igmV28Ik+swxApNDI7OjFyKyHsWFyo1k11rrCIHpgdH5rcMmzlFsqNzdjBV4kvhVFmUUE8WHyh3lNHqvvMDkoMjs0YET9yhOVO5Oxgo8QfsKCzKBB/8LvivMp7TA5IDI7AipdtegeFG5u/gq8CDp0xdjEMvDaxQvKjea36N6ZQUmB0Nm71qMFTkUNyp3BcP7miUFJgdCZu9YiJUZFDsqdyXLt9oK3En55GXoRDF0nOJH5Q5B6rhUTmByEGR2x0xTH6UYUrmrh6HAjYRPXYDG54ceo1hSuaHw7oqVEpgcAJm9cgGI2hRLKncl4zICk/DJ7JXDJ2tTTKncVawVuIHsaUNvePLyIxRTKncV0BICk9DJ7FVDz1KXYkvlruCuwF9QPWnQK5ZntibFl8qd5fXZ/fQCv8Ke+fHy/WX4+kmDHoaw+CLFmMqNxple4NkHz/wCMCP/bN9XuU+JROVGz/V4gaOBWS+eACUTlRtJUIEjaVpriAAlEpU7BOnBJQWOpGmtYQKUTFTuMKi7iwocRdI6UwQokajcKVjvLitwFEnrTBOgZKJyp4HdbjcFjqBojRAClEhUbgQ0BY6gaI0wApRMVO4sOAWeJej9UAKUSFTuLDwFniXo/XAClExU7gxABZ6h590lBCiRqNwZiAo8Q8+7ywhQMlG5oyAVeJSc95YSoESickdhKvAoOe8tJ0DJROWOAFXgEWre2UKAEonKHYGqwCPUvLONACUTldsLVoF7iXl+KwFKJCq3F64C9xLz/HYClExUbg9gBe6h5VmEACUSldsDWYF7aHkWI0DJROW2glbgVlKeQwm8ikT9yPz/RlNgaivMlUAAAQUOgGgJCVAEFJgib64EAggocABES0iAIqDAFHlzJRBAQIEDIFpCAhQBBabImyuBAAIKHADREhKgCCgwRd5cCQQQUOAAiJaQAEVAgSny5koggIACB0C0hAQoAgpMkTdXAgEEFDgAoiUkQBFQYIq8uRIIIKDAARAtIQGKgAJT5M2VQAABBQ6AaAkJUAQUmCJvrgQCCChwAERLSIAioMAUeXMlEEBAgQMgWkICFAEFpsibK4EAAgocANESEqAIKDBF3lwJBBBQ4ACIlpAARUCBKfLmSiCAgAIHQLSEBCgCCkyRN1cCAQQUOACiJSRAEVBgiry5EgggoMABEC0hAYqAAlPkzZVAAAEFDoBoCQlQBBSYIm+uBAIIKHAAREtIgCKgwBR5cyUQQECBAyBaQgIUAQWmyJsrgQAC/wKFIpBbbILH5gAAAABJRU5ErkJggg==)

gfwilliams, 2020-05-27:
Ok, so I'm looking at this and I can make the poly fill more reliable, but then it effectively cuts off the top pixel.

This is the poly fill from https://alienryderflex.com/polygon_fill/ but in JS:
[image]

```
g.clear();

function fillPoly(a) {
  var polyCorners = 0;
  var polyX=[],polyY=[];
  for (var i=0;i<a.length;i+=2) {
    polyX[polyCorners]=a[i];
    polyY[polyCorners]=a[i+1];
    polyCorners++;
  }
  
  var  nodes, pixelX, pixelY, i, j, swap ;

  //  Loop through the rows of the image.
  for (pixelY=0; pixelY<30; pixelY++) {
    var nodeX=[];
  
    //  Build a list of nodes.
    nodes=0; j=polyCorners-1;
    for (i=0; i<polyCorners; i++) {
      if (polyY[i]<pixelY && polyY[j]>=pixelY
      ||  polyY[j]<pixelY && polyY[i]>=pixelY) {
        nodeX[nodes++]=0|(polyX[i]+(pixelY-polyY[i])/(polyY[j]-polyY[i])
        *(polyX[j]-polyX[i])); }
      j=i; }
    //  Sort the nodes, via a simple "Bubble" sort.
    i=0;
    while (i<nodes-1) {
      if (nodeX[i]>nodeX[i+1]) {
        swap=nodeX[i]; nodeX[i]=nodeX[i+1]; nodeX[i+1]=swap; if (i) i--; }
      else {
        i++; }}
    console.log(pixelY,nodeX);

    //  Fill the pixels between node pairs.
    for (i=0; i<nodes; i+=2) {
        for (pixelX=nodeX[i]; pixelX<nodeX[i+1]; pixelX++) 
          g.fillRect(pixelX*4,pixelY*4,pixelX*4+3,pixelY*4+3); }}
}

fillPoly([ 16, 0, 31, 31, 26, 31, 16, 12, 6, 28, 0, 27 ]);
```

It's not quite as good as that in reality due to integer arithmetic, but because of the `polyY[i]<pixelY` you're never filling up to the top pixel. I don't know - maybe that's ok.

One option is to offset what we check against, so effectively ` for (pixelY=0.5;` rather than ` for (pixelY=0; `...

closed 2020-05-27 by gfwilliams

nebbishhacker, 2020-05-29:
I wonder if it might still be worth adding that half-pixel offset to what we check against. Right now fillPoly effectively sample from the top-left corner of each pixel. Sampling from the center of each pixel would make intuitive sense to me, and it would match how I've been assuming things would be rendered while working on the font in Inkscape.

Aesthetically, it might improve the appearance of gentle slopes. As of https://github.com/espruino/Espruino/commit/5d61b6757e3687030a2ea6adee8afdf2fa641bda the poly `[1, 1, 30, 2, 30, 4, 1, 3]` renders as:

```
................................
.#..............................
.#############################..
.#############################..
................................
```

With a half-pixel offset, it would render as:

```
................................
.##############.................
.#############################..
...............###############..
................................
```

which looks more "correct" to my eyes.

gfwilliams, 2020-05-29:
That sounds like a good plan to me,

So just to be sure you mean basically subtracting 8 here: https://github.com/espruino/Espruino/blob/5d61b6757e3687030a2ea6adee8afdf2fa641bda/libs/graphics/graphics.c#L438

nebbishhacker, 2020-05-29:
Yes, that is precisely what I mean, for both vx and vy.

gfwilliams, 2020-05-29:
Thanks - just done. Rendering looks a whole lot more sensible now!

commit 42bc8343 (2020-05-27): Graphics.fillPoly now uses a more normal fill algorithm that doesn't attempt to fill to the top right (fix #1796)