espruino/Espruino #1804: ESP32D0WDQ6 (revision 1) not flashing correctly since release 2.01

stefanodellavalle, 2020-04-23:
I've tested every combination of cables, esptool parameters and options, but I can't flash a ESP32 dev board correctly with higher versions than the 2.0

Version 2.0 works perfectly.

Flashing younger versions produce a boot loop. Attached a screen capture.

[image]

MaBecker, 2020-04-23:
Erase flash and try again.

Please share information about to board you use:

    esptool.py --port $COMPORT flash_id

output should look similar to this one

```
esptool.py v2.8
Serial port /dev/cu.SLAB_USBtoUART
Connecting........___
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a4:cf:12:76:64:b4
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

MaBecker, 2020-04-23:
ESP32-S0WD and ESP32-D2WD can only run with 160 MHZ

stefanodellavalle, 2020-04-23:
Yes, every test on every version has been done erasing the flash and then flashing the code.

Here is the output.

esptool.py --port COM4 flash_id
esptool.py v2.8
Serial port COM4
Connecting........_____....._
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 160MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 84:0d:8e:11:2f:94
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

MaBecker, 2020-04-24:
The cpu frequency is part of the esp-idf configuration and is set to 240 MHz.
> 40MHz/80MHz is the SPI flash frequency setting. 
It is not related to CPU frequency setting.
CPU frequency is set at the time this library is compiled,
and can not be changed afterwards.

If you want espruino on your board you have to go through the esp-idf setup process and create libs for 160 MHz. With those libs you can build a ESP32 160 MHz Espruino version.

MaBecker, 2020-04-24:
> Chip is ESP32D0WDQ6 (revision 1)
Crystal is 40MHz

same

> Features: WiFi, BT, Dual Core, 160MHz, VRef calibration in efuse, Coding Scheme None

different

Is there a way to change 160 to 240 MHz device wise?

MaBecker, 2020-04-24:
> Version 2.0 works perfectly.

I guess after that there where a switch from 160 to 240 MHz.

sdellava, 2020-04-24:
Ok. So I've to consider that this device will be not supported from higher release?

In the board there no way to set the frequency.

MaBecker, 2020-04-24:
> Ok. So I've to consider that this device will be not supported from higher release?

Or build by your own, as mentioned above. To be honest not easy but not impossible.;-)

sdellava, 2020-04-24:
Ok. well, it seems very much easier to buy a newer board :) 

Thanks for your support. To me this issues can be closed.

MaBecker, 2020-04-24:
Ok, yes it is very difficult to pick a proper ESP32 board, there are to many.

Maybe you like to  post a forum conversation and ask other users  to name a recommendation for a board they use or tried with or without success.

closed 2020-04-24 by MaBecker

AkosLukacs, 2020-04-25:
Oh, wouldn't it be nice to gather the working or not-working boards "[somewhere](https://www.espruino.com/ESP32)"? With date, board revision/version. Ok, might be asking too much...

MaBecker, 2020-04-25:
> Oh, wouldn't it be nice to gather the working or not-working boards "[somewhere](https://www.espruino.com/ESP32)"? With date, board revision/version. Ok, might be asking too much...

Sure, just go and updare that source on [githup](https://github.com/espruino/EspruinoDocs/blob/master/boards/ESP32.md)

maybe in section Limitation

dr2man, 2024-12-13:
what development board should I select from the Arduino IDE