espruino/Espruino #1805: How to build PICO firmware files for WIZNet board?

MaBecker, 2020-04-24:
Not sure how to build a espruino_2v05_pico_1r3_wiznet.bin?

Here I find some information https://www.espruino.com/WIZnet

    BOARD=PICO_R1_3 WIZNET=1 RELEASE=1 ESPRUINO_1V3=1 make

output

    make/sanitycheck.make:12: *** You now need to build with 'BOARD=ESPRUINOBOARD make'.  Stop.

tried 

     BOARD=PICO_R1_3 RELEASE=1  WIZNET=1 make

creates this files

    espruino_2v05.19_pico_1r3.elf
    espruino_2v05.19_pico_1r3.bin
    espruino_2v05.19_pico_1r3.lst

how to create this file?
    espruino_2v05_pico_1r3_wiznet.bin

gfwilliams, 2020-04-24:
Do the builds from https://www.espruino.com/Download#pico work ok for you?

> creates this files

I think those are the correct files. Did you test them? They're just not named as you'd expect - that's done by `create_zip` afaik which creates the distribution zip.

MaBecker, 2020-04-24:
> Do the builds from https://www.espruino.com/Download#pico work ok for you?

Yes it works.

After extending my build command like this 

    PAD_FOR_BOOTLOADER=1 WIZNET=1 RELEASE=1 make

It work's like a charm!

closed 2020-04-24 by MaBecker