espruino/Espruino #1806: Any plans to extend the WIZnet550 driver?

MaBecker, 2020-04-24:
Would be cool to have 

* getHostname()  and setHostname() 

* reset() to release ip given with setIP() only for WS550 because it has a default ip after reset/powerup

* ......

To Be extended ...

gfwilliams, 2020-04-24:
> getHostname

If the hostname is actually configurable, then yes. I'm not sure if the driver does mDNS, so if it doesn't there's be no point having the hostname.

> reset() 

What's the use for this one? Can you not just call setIP with `0,0,0,0`?

One thing I've started doing is adding a function that allows the user to send command straight to the chip. Might be a way to allow some of the more interesting functionality to be exposed...

MaBecker, 2020-04-26:
@opichals do you have any whishes or needs to enhance WIZnet lib?

MaBecker, 2020-04-26:
> I'm not sure if the driver does mDNS

Look's like mDNS and uses a fixed hostname.

This is hows it could be [done](https://github.com/arduino-libraries/Ethernet/pull/50/commits/a45bb068120e1e5bdf7943222373f3ad8076ba68#diff-961b3f6679dc8b161750718d975b9747). 

> What's the use for this one? Can you not just call setIP with 0,0,0,0?

Some boards have a reset pin and if there is a reset they startup with a default like  this 

so just a optional pin {rst: 
WIZ550io

```
={
  ip: "192.168.1.2",
  subnet: "255.255.255.0",
  gateway: "192.168.1.1",
  dns: "0.0.0.0",
  mac: "00:08:dc:57:c1:f0"
 }
```

Ok, for other board a setIP({values}) is ok.

gfwilliams, 2020-04-27:
Ahh, So I'm not sure that's mDNS - it just gives a hostname when it configures itself via DHCP? That sounds like a great thing to add.

I'm not too sure about the reset - for the cases where you need it, can't you just toggle the reset pin yourself?

MaBecker, 2020-04-30:
What about adding this too?

* ping(cb) 

* getState()   return value of JsNetworkState as string

gfwilliams, 2020-04-30:
That'd be cool if you want to add it.

Although with ping - maybe be careful it doesn't add too much extra size to the binary?

MaBecker, 2020-05-29:
```
>eth.getStatus()
={
  state: "online",
  id: "W5500",
  phylink: "up",
  phypowmode: "normal"
 }
> 
```

MaBecker, 2020-07-01:
calling `eth.setIP()`has some extra jsiConsolePrintf() like:

https://github.com/espruino/Espruino/blob/3b74efba495fbdef3be76dda238a8208b15aacf2/libs/network/wiznet/DHCP/dhcp.c#L163

https://github.com/espruino/Espruino/blob/3b74efba495fbdef3be76dda238a8208b15aacf2/libs/network/wiznet/DHCP/dhcp.c#L291

and so on...

I can't remember which function to use so those are not printed when build with `RELEASE=1`.

gfwilliams, 2020-07-02:
That sounds good - there isn't actually a specific function for debug messages, so `ifndef RELEASE` is probably the best bet

closed 2020-07-03 by MaBecker