espruino/Espruino #1807: SPIFLASH breaks peek

fanoush, 2020-04-27:
I tried enabling  SPIFLASH in board file with P8 nrf52 smartwatch and it seems to work fine but by enabling it I locked myself out of  NRF custom chips access due to peek32 not working for all addresses >0x40000000. Had to upload InlineC code with alternative peek/poke implementation to get it working :-) Not sure if poke was broken too, maybe not when looking at source now.

There are 3 reasons, not sure which one (or all) you want to fix.
- SPIFLASH base address 0x40000000 with no upper bound checking conflicts with all NRF custom registers, moving to 0x6 would not conflict with NRF (but arm cortex 0xE... area  would be still broken)
- peek calls [jshFlashGetMemMapAddress](https://github.com/espruino/Espruino/blob/e78ba062ca299420e3ffebb3527c357b72d8db49/src/jswrap_io.c#L106) which returns 0 for SPIFLASH area so any peek works on address 0 - not sure if this call is needed and what is the reason, should poke/peek write/read to external spi flash? should poke implementation call  jshFlashGetMemMapAddress in jswrap_io_poke as well like jswrap_io_peek does?
- there are no checks for end of flash anywhere including [jshFlashGetMemMapAddress implementation](https://github.com/espruino/Espruino/blob/f5f614864aa47f2af7df34a4e0692935f6984314/targets/nrf5x/jshardware.c#L1917) perhaps `ptr >= SPIFLASH_BASE && ptr < (SPIFLASH_BASE+SPIFLASH_LENGTH)` would be better there? There are many places checking `addr` in same way there too but maybe this was omitted for performance/size reason?

For now I kept jshFlashGetMemMapAddress call there, moved SPIFLASH base address to 0x60000000 and added checks for end of flash.

Maybe just removing jshFlashGetMemMapAddress call from jswrap_io_peek would be enough? However there is some ESP related comment.

gfwilliams, 2020-04-27:
I think the best bet would be to move the area and maybe add bounds checking. What about just moving to `0xF0000000` though?

> Maybe just removing jshFlashGetMemMapAddress call from jswrap_io_peek would be enough?

Yeah, that's done for ESP8266/etc. Probably best not to change it...

fanoush, 2020-04-27:
>  What about just moving to 0xF0000000 though?

Possibly, it could be also platform dependent as different chips have different unused areas. What about moving it to board file so it is near [this](https://github.com/espruino/Espruino/blob/master/boards/BANGLEJS.py#L70). Maybe directly to SPIFLASH specification like size and provide no default (like size)

> Yeah, that's done for ESP8266/etc. Probably best not to change it

Can be ifdef-ed for ESP though, current breakage is because jshFlashGetMemMapAddress has different meaning for esp and nrf52(?), you are using it with 0 value as bool [here](https://github.com/espruino/Espruino/blob/96248cbb1440e5d8d15ee1f94eefea9ab616c070/src/jsflash.c#L523) which breaks peek semantics(?) - reading from address 0. Or if reading address 0 is invalid like NULL pointer there should be check and exception(?). Now it reads wrong memory instead. Adding  something like jshFlashIsDirectlyMappedMem would fit better your SPI flash use case without breaking jshFlashGetMemMapAddress which should be no-op 1:1 translation for peek (unless you want peek/poke to also read/write spi flash behind scenes)

gfwilliams, 2020-04-27:
Just made it configurable and added and address checks - it made sense to stick it near the `spiflash` stuff.

But yeah, having jshFlashIsDirectlyMappedMem seems like a good idea.

fanoush, 2020-04-27:
Excellent, thank you!  Can be closed now(?)

gfwilliams, 2020-04-27:
Yeah, I guess so. peek won't work but I think that's ok for now?

closed 2020-04-27 by gfwilliams