espruino/Espruino #1813: IDE enhancement: save log & allow select in left pane

FransM, 2020-05-05:
It would be nice if it was possible to select text in the left pane of the IDE so it can e.g. be cut into an email message.
Also (or alternately) it would be nice if the commands given could be saved to a logfile

gfwilliams, 2020-05-05:
> It would be nice if it was possible to select text in the left pane of the IDE so it can e.g. be cut into an email message.

As was mentioned in a reply to you on the forum, this is already possible and works fine.

Just click and drag and the text is immediately copied to the terminal.

gfwilliams, 2020-05-05:
Also, there's already an issue for saving to a file: https://github.com/espruino/EspruinoWebIDE/issues/225

closed 2020-05-05 by gfwilliams

fanoush, 2020-05-05:
> Just click and drag and the text is immediately copied to the terminal.

which is a bit confusing at first, but works. one doesn't need to press ctrl+c or find edit->copy menu, which is good, but still unexpected. took me some time to figure out why the selection is vanishing immediatelly and that the fature indeed works. maybe some feedback like android toast message 'copied'

gfwilliams, 2020-05-05:
> maybe some feedback like android toast message 'copied' would help

Yeah, that could be good I guess.

The lack of using Ctrl-C is intentional, because Ctrl-C breaks out of running code or deletes the current line of text when in the terminal - I had to find a way of doing it that didn't involve that :)