espruino/Espruino #1820: IIlegal memory access may lead to arbitrary memory write inside jsvGarbageCollectMarkUsed

vulsReport, 2020-05-09:
Hello,
I found that IIlegal memory access may lead to arbitrary memory write inside jsvGarbageCollectMarkUsed . When Object property name is marked ，it will be regarfed as nativeStr struct. And read content pointed by property name and marked，which will result arbitrary memory write.
Please confirm~~
poc is here:
[espruino1.js.zip](https://github.com/espruino/Espruino/files/4602706/espruino1.js.zip)

test version:
commit d543731 (HEAD -> master, origin/master, origin/HEAD) (20200505)

environment

gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
run ./espruino poc
(gdb) r
Starting program: /home/zdz/Espruino/espruino /home/zdz/debugBug/espruino1.js
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6e85700 (LWP 4410)]

| |_ ___ ___ _ ||___ ___
| | -| . | | | | | | . |
||| || |||||_|
|| espruino.com
2v05.41 (c) 2019 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

{ }

Thread 1 "espruino" received signal SIGSEGV, Segmentation fault.
jsvIsString (
v=<error reading variable: Cannot access memory at address 0x7fffff7feff0>) at src/jsvar.c:73
73 bool jsvIsString(const JsVar *v) { return v && (v->flags&JSV_VARTYPEMASK)>=_JSV_STRING_START && (v->flags&JSV_VARTYPEMASK)<=_JSV_STRING_END; } ///< String, or a NAME too
(gdb) bt
#0 jsvIsString (
v=<error reading variable: Cannot access memory at address 0x7fffff7feff0>) at src/jsvar.c:73
#1 0x0000555555565728 in jsvHasCharacterData (v=0x7ffff6644610)
at src/jsvar.c:384
#2 0x000055555556f8d2 in jsvGarbageCollectMarkUsed (var=0x7ffff6644610)
at src/jsvar.c:3715
#3 0x000055555556f97a in jsvGarbageCollectMarkUsed (var=0x7ffff6644650)
at src/jsvar.c:3730
#4 0x000055555556f9cb in jsvGarbageCollectMarkUsed (var=0x7ffff6644670)
at src/jsvar.c:3738

Dongzhuo Zhao working with ADLab of Venustech

vulsReport, 2020-05-09:
#1818 and #1819 reported by Dongzhuo Zhao .Because that account was flagged,so I report this issue again. Please reply this. Look forward to your reply! Thanks~~

gfwilliams, 2020-05-11:
Thanks - I do get a segfault, but there's a massive stack trace which seems to be caused by creating an almost 100,000 element deep recursive structure.

If so that's this issue: https://github.com/espruino/Espruino/issues/1765

If you change `new Object(v2)` to `new Object()` then it appears to work just fine.

Could you maybe try and reproduce the problem without the recursion?

gfwilliams, 2020-05-27:
Closing this as it's a duplicate of #1765 by the look of it, and no response.

closed 2020-05-27 by gfwilliams